google-deepmind/dm_control #123: EGL initialization error

ajabri, 2019-12-29:
Hi,

I've recently come across the following error, on a machine with Nvidia driver version: 440.33.01    CUDA version: 10.2, when trying to use EGL for headless rendering. The same exact code was running properly on a machine with CUDA Version 10.1 and driver version 390.

Any idea how to fix the issue? I've tried reinstalling dm_control with pip.
```
  File "/home/aj/miniconda3/envs/rlpyt/lib/python3.7/site-packages/dm_control/suite/__init__.py", line 28, in <module>
    from dm_control.suite import acrobot
  File "/home/aj/miniconda3/envs/rlpyt/lib/python3.7/site-packages/dm_control/suite/acrobot.py", line 24, in <module>
    from dm_control import mujoco
  File "/home/aj/miniconda3/envs/rlpyt/lib/python3.7/site-packages/dm_control/mujoco/__init__.py", line 18, in <module>
    from dm_control.mujoco.engine import action_spec
  File "/home/aj/miniconda3/envs/rlpyt/lib/python3.7/site-packages/dm_control/mujoco/engine.py", line 44, in <module>
    from dm_control import _render
  File "/home/aj/miniconda3/envs/rlpyt/lib/python3.7/site-packages/dm_control/_render/__init__.py", line 75, in <module>
    Renderer = import_func()
  File "/home/aj/miniconda3/envs/rlpyt/lib/python3.7/site-packages/dm_control/_render/__init__.py", line 36, in _import_egl
    from dm_control._render.pyopengl.egl_renderer import EGLContext
  File "/home/aj/miniconda3/envs/rlpyt/lib/python3.7/site-packages/dm_control/_render/pyopengl/egl_renderer.py", line 66, in <module>
    EGL_DISPLAY = create_initialized_headless_egl_display()
  File "/home/aj/miniconda3/envs/rlpyt/lib/python3.7/site-packages/dm_control/_render/pyopengl/egl_renderer.py", line 49, in create_initialized_headless_egl_display
    for device in EGL.eglQueryDevicesEXT():
  File "/home/aj/miniconda3/envs/rlpyt/lib/python3.7/site-packages/dm_control/_render/pyopengl/egl_ext.py", line 65, in eglQueryDevicesEXT
    success = _eglQueryDevicesEXT(max_devices, devices, num_devices)
ctypes.ArgumentError: argument 2: <class 'TypeError'>: expected LP_c_void_p instance instead of EGLDeviceEXT_pointer_Array_10
```

Thanks,
A

xinleipan, 2020-01-02:
Try to do

$ export DISPLAY=:0

in your terminal and try again, it works for me.

ajabri, 2020-01-03:
@xinleipan In this case, I'm trying to use EGL for rendering. Setting "DISPLAY=:0" results in the below error. Are you rendering with glfw or osmesa?

```
Traceback (most recent call last):
  File "main_maw.py", line 6, in <module>
    import dmc_wrapper as dmc2gym
  File "/home/aj/rlpyt/selfish/dmc_wrapper.py", line 2, in <module>
    from dm_control import suite
  File "/home/aj/miniconda3/envs/rlpyt/lib/python3.7/site-packages/dm_control/suite/__init__.py", line 28, in <module>
    from dm_control.suite import acrobot
  File "/home/aj/miniconda3/envs/rlpyt/lib/python3.7/site-packages/dm_control/suite/acrobot.py", line 24, in <module>
    from dm_control import mujoco
  File "/home/aj/miniconda3/envs/rlpyt/lib/python3.7/site-packages/dm_control/mujoco/__init__.py", line 18, in <module>
    from dm_control.mujoco.engine import action_spec
  File "/home/aj/miniconda3/envs/rlpyt/lib/python3.7/site-packages/dm_control/mujoco/engine.py", line 44, in <module>
    from dm_control import _render
  File "/home/aj/miniconda3/envs/rlpyt/lib/python3.7/site-packages/dm_control/_render/__init__.py", line 67, in <module>
    Renderer = import_func()  # pylint: disable=invalid-name
  File "/home/aj/miniconda3/envs/rlpyt/lib/python3.7/site-packages/dm_control/_render/__init__.py", line 36, in _import_egl
    from dm_control._render.pyopengl.egl_renderer import EGLContext
  File "/home/aj/miniconda3/envs/rlpyt/lib/python3.7/site-packages/dm_control/_render/pyopengl/egl_renderer.py", line 68, in <module>
    raise ImportError('Cannot initialize a headless EGL display.')
ImportError: Cannot initialize a headless EGL display.
```

xinleipan, 2020-01-04:
I'm not sure, sorry. Maybe it's GLFW, I just installed MuJoCo normally with a Ubuntu 16.04 machine, CUDA 8.0 and NVIDIA GPU.

episodeyang, 2020-01-04:
same issue here on a ubuntu box.

episodeyang, 2020-01-04:
Problem solved, use this: @ajabri 

```
export MUJOCO_GL="glfw"
```

In order to avoid the EGL, one way I have seen working is to use the `GLFW` backend. To do so, you need to set the environment flag to `glfw`. Then you need a display port that is actually available.

ajabri, 2020-01-04:
I need headless rendering, so I need to use EGL as my rendering backend.
The same code worked on another machine, the only difference being (I
believe) that the machine on which I obtain the above error has a much more
recent nvidia driver (440 vs 390).

episodeyang, 2020-01-04:
@ajabri if X display is not available, you can use "osmesa" backend. 

(for the interest of others: won't work even if you install Xvfb and GLFW because they require X-display)

```
export MUJOCO_GL="osmesa"
```

Here is the full setup for a headless linux box with no built-in X display:

```bash
      export MUJOCO_GL=osmesa
      export MJLIB_PATH=$HOME/.mujoco/mujoco200/bin/libmujoco200.so
      export MJKEY_PATH=$HOME/.mujoco/mujoco200/mjkey.txt
      export LD_LIBRARY_PATH=$HOME/.mujoco/mujoco200/bin:$LD_LIBRARY_PATH
      export MUJOCO_PY_MJPRO_PATH=$HOME/.mujoco/mujoco200/
      export MUJOCO_PY_MJKEY_PATH=$HOME/.mujoco/mujoco200/mjkey.txt
```

@saran-t maybe we can add this to the documentation? :)

ajabri, 2020-01-04:
Hmmm, EGL should allow for hardware accelerated headless rendering without
X display. Has worked worked for me in the past.

episodeyang, 2020-01-04:
Maybe we can ping @saran-t about that

lich14, 2020-01-05:
> @ajabri if X display is not available, you can use "osmesa" backend.
> 
> (for the interest of others: won't work even if you install Xvfb and GLFW because they require X-display)
> 
> ```
> export MUJOCO_GL="osmesa"
> ```
> 
> Here is the full setup for a headless linux box with no built-in X display:
> 
> ```shell
>       export MUJOCO_GL=osmesa
>       export MJLIB_PATH=$HOME/.mujoco/mujoco200/bin/libmujoco200.so
>       export MJKEY_PATH=$HOME/.mujoco/mujoco200/mjkey.txt
>       export LD_LIBRARY_PATH=$HOME/.mujoco/mujoco200/bin:$LD_LIBRARY_PATH
>       export MUJOCO_PY_MJPRO_PATH=$HOME/.mujoco/mujoco200/
>       export MUJOCO_PY_MJKEY_PATH=$HOME/.mujoco/mujoco200/mjkey.txt
> ```
> 
> @saran-t maybe we can add this to the documentation? :)

This way exactly solve me from those terrible mistakes

saran-t, 2020-01-05:
I can't see how a CUDA upgrade could have changed anything given that we are specifying the argument types ourselves, but one thing that you could try is to change this line

https://github.com/deepmind/dm_control/blob/master/dm_control/_render/pyopengl/egl_ext.py#L65

to

`success = _eglQueryDevicesEXT(max_devices, ctypes.POINTER(ctypes.c_void_p)(ctypes.addressof(devices)), num_devices)`

Let me know if this works.

ajabri, 2020-01-06:
The above change gives me the following error:

```
Traceback (most recent call last):
  File "main_maw.py", line 6, in <module>
    import dmc_wrapper as dmc2gym
  File "/home/aj/rlpyt/selfish/dmc_wrapper.py", line 2, in <module>
    from dm_control import suite
  File "/home/aj/miniconda3/envs/rlpyt/lib/python3.7/site-packages/dm_control/suite/__init__.py", line 28, in <module>
    from dm_control.suite import acrobot
  File "/home/aj/miniconda3/envs/rlpyt/lib/python3.7/site-packages/dm_control/suite/acrobot.py", line 24, in <module>
    from dm_control import mujoco
  File "/home/aj/miniconda3/envs/rlpyt/lib/python3.7/site-packages/dm_control/mujoco/__init__.py", line 18, in <module>
    from dm_control.mujoco.engine import action_spec
  File "/home/aj/miniconda3/envs/rlpyt/lib/python3.7/site-packages/dm_control/mujoco/engine.py", line 44, in <module>
    from dm_control import _render
  File "/home/aj/miniconda3/envs/rlpyt/lib/python3.7/site-packages/dm_control/_render/__init__.py", line 67, in <module>
    Renderer = import_func()  # pylint: disable=invalid-name
  File "/home/aj/miniconda3/envs/rlpyt/lib/python3.7/site-packages/dm_control/_render/__init__.py", line 36, in _import_egl
    from dm_control._render.pyopengl.egl_renderer import EGLContext
  File "/home/aj/miniconda3/envs/rlpyt/lib/python3.7/site-packages/dm_control/_render/pyopengl/egl_renderer.py", line 66, in <module>
    EGL_DISPLAY = create_initialized_headless_egl_display()
  File "/home/aj/miniconda3/envs/rlpyt/lib/python3.7/site-packages/dm_control/_render/pyopengl/egl_renderer.py", line 49, in create_initialized_headless_egl_display
    for device in EGL.eglQueryDevicesEXT():
  File "/home/aj/miniconda3/envs/rlpyt/lib/python3.7/site-packages/dm_control/_render/pyopengl/egl_ext.py", line 66, in eglQueryDevicesEXT
    success = _eglQueryDevicesEXT(max_devices, ctypes.POINTER(ctypes.c_void_p)(ctypes.addressof(devices)), num_devices)
TypeError: expected c_void_p instead of int
```

alimuldal, 2020-01-06:
I think this is was due to an upstream change in PyOpenGL. It seems that in versions newer than 3.1.4, `OpenGL.EGL` has its own `EGLDeviceEXT` member (https://github.com/mcfletch/pyopengl/commit/38f4cc533b2ed3e85c585c4a4f00ed06f7aa6056), which overrides our assignment of `EGLDeviceEXT = ctypes.c_void_p` when we do the wildcard import on the last line. Consequently we end up with mismatching pointer types in `_eglQueryDevicesEXT` and in the body of `eglQueryDevicesEXT`. I'll put together a fix.

closed 2020-01-07 by alimuldal

ajabri, 2020-01-08:
Thank you @alimuldal!

bdtacchi, 2022-03-21:
So I just spent hours on this, and I'm not an expert, so I might say something wrong. But, in my case, I was trying to run it over SSH, so I suspect that's what caused the headless display to not work. So, I tried switching the backend, but `export MUJOCO_GL="glfw"` or osmesa or whatever didn't make it not choose egl, so I had to go to the `__init.py__` file from `dmcontrol/_render` and set the `BACKEND` variable right at the beginning to `'osmesa'`, which then made it finally use osmesa as the backend.

balloch, 2024-06-28:
It is worth noting that I am able to make this error go away when it comes up by restarting the machine. This is not a long term solution, but it means that whatever is causing it-probably something to do with the EGL backend state-is sensitive to machine state.

shuoyang2000, 2025-01-01:
I found this document very useful: https://pytorch.org/rl/stable/reference/generated/knowledge_base/MUJOCO_INSTALLATION.html

There are 15 "Common Issues during import or when rendering Mujoco Environments".

It's better to have sudo access, otherwise conda may not work.

commit 0b7369e5 (2020-01-07): Fix a bug in `egl_ext.py` due to mismatching `EGLDeviceExt` pointer types

As of PyOpenGL v3.1.4, `OpenGL.EGL` has its own `EGLDevicesEXT` member, which overrides our assignment of `EGLDeviceEXT = ctypes.c_void_p` when we later do `from OpenGL.EGL import *`. Consequently we end up with mismatching pointer types in the `CFUNCTYPE` wrapper for `_eglQueryDevicesEXT` and in the body of `eglQueryDevicesEXT`. I have bumped the required version of PyOpenGL to 3.4.1 and switched to using `EGL.EGLDevicesEXT` instead of `ctypes.c_void_p`.

Closes #123

PiperOrigin-RevId: 288398964
Change-Id: If5b1a9dc92 [... cut]