google-deepmind/dm_control #125: change visual appearance of rendered environment?

xinleipan, 2020-01-04:
I found that in some of the xml files there is no parameters for defining the color of objects, or the size of objects. However, in the original MuJoCo environment, just change the xml file can do this.  Is there any quick way to change the visual appearance of the environments such as object color and size?

alimuldal, 2020-01-06:
The recommended way to change the colors or sizes of geoms is to edit the XML files (see the [official MuJoCo docs](http://mujoco.org/book/XMLreference.html) for more details on the MJCF schema).

It is also possible to modify colors by directly writing into fields in the `physics.model` struct after compilation (we use this technique for [visualizing per-step reward values in the control suite](https://github.com/deepmind/dm_control/blob/8deb836372ad6a3810dd945143c1a937d6fe9c27/dm_control/suite/base.py#L105) in the control suite). I would not recommend changing geom sizes after compilation, though. MuJoCo precomputes various things that depend on geom size when the model is compiled (e.g. bounding boxes), so modifying the geom sizes after compilation can result in aberrant simulation behavior.

closed 2020-01-06 by alimuldal

geyang, 2020-05-20:
Here is an example

```python
import gym
import numpy as np
from dm_control import suite
import matplotlib.pyplot as plt

suite.ALL_TASKS

# env = gym.make("dm2gym:HopperHop-v0")
env = gym.make("dm2gym:ManipulatorBring_peg-v0")
env.reset()
img = env.render('rgb_array', width=240, height=240)
plt.imshow(img)
```

[image]

```python
env = gym.make("dm2gym:ManipulatorInsert_peg-v0")
env.reset()
env.step(np.zeros(5))
img = env.render('rgb_array', width=240, height=240)
plt.imshow(img)
env.step(np.zeros(5))
img = env.render('rgb_array', width=240, height=240)
plt.imshow(img)
```

[image]

```python
env.step(np.zeros(5))
img = env.render('rgb_array', width=240, height=240)
plt.imshow(img)
```

[image]

```python
env.env.env.physics.named.model.mat_rgba
```

    FieldIndexer(mat_rgba):
                            r         g         b         a         
     0               grid [ 1         1         1         1       ]
     1               self [ 0.7       0.5       0.3       1       ]
     2       self_default [ 0.7       0.5       0.3       1       ]
     3     self_highlight [ 0         0.5       0.3       1       ]
     4           effector [ 0.7       0.4       0.2       1       ]
     5   effector_default [ 0.7       0.4       0.2       1       ]
     6 effector_highlight [ 0         0.5       0.3       1       ]
     7         decoration [ 0.3       0.5       0.7       1       ]
     8                eye [ 0         0.2       1         1       ]
     9             target [ 0         0         0         0       ]
    10     target_default [ 0.6       0.3       0.3       1       ]
    11   target_highlight [ 0         0         0         0       ]
    12               site [ 0.5       0.5       0.5       0.3     ]
    13         background [ 1         1         1         1       ]

```python
env.env.env.physics.named.model.mat_rgba["decoration"][:] = 0
```

```python
img = env.render('rgb_array', width=240, height=240)
plt.imshow(img)
```

    <matplotlib.image.AxesImage at 0x13757ff60>

[image]

```python
env.env.env.physics.named.model.geom_rgba
```

    FieldIndexer(geom_rgba):
                       r         g         b         a         
     0         floor [ 0.5       0.5       0.5       1       ]
     1         wall1 [ 0.5       0.5       0.5       1       ]
     2         wall2 [ 0.5       0.5       0.5       1       ]
     3    background [ 0.5       0.5       0.5       1       ]
     4      arm_root [ 0.5       0.5       0.5       1       ]
     5     upper_arm [ 0.5       0.5       0.5       1       ]
     6    middle_arm [ 0.5       0.5       0.5       1       ]
     7     lower_arm [ 0.5       0.5       0.5       1       ]
     8          hand [ 0.5       0.5       0.5       1       ]
     9         palm1 [ 0.5       0.5       0.5       1       ]
    10         palm2 [ 0.5       0.5       0.5       1       ]
    11        thumb1 [ 0.5       0.5       0.5       1       ]
    12        thumb2 [ 0.5       0.5       0.5       1       ]
    13     thumbtip1 [ 0.5       0.5       0.5       1       ]
    14     thumbtip2 [ 0.5       0.5       0.5       1       ]
    15       finger1 [ 0.5       0.5       0.5       1       ]
    16       finger2 [ 0.5       0.5       0.5       1       ]
    17    fingertip1 [ 0.5       0.5       0.5       1       ]
    18    fingertip2 [ 0.5       0.5       0.5       1       ]
    19         blade [ 0.5       0.5       0.5       1       ]
    20         guard [ 0.5       0.5       0.5       1       ]
    21        pommel [ 0.5       0.5       0.5       1       ]
    22        slot_0 [ 0.5       0.5       0.5       1       ]
    23        slot_1 [ 0.5       0.5       0.5       1       ]
    24        slot_2 [ 0.5       0.5       0.5       1       ]
    25  target_blade [ 0.5       0.5       0.5       1       ]
    26  target_guard [ 0.5       0.5       0.5       1       ]
    27 target_pommel [ 0.5       0.5       0.5       1       ]

```python
env.env.env.physics.named.model.geom_rgba["blade"][:] = 0
img = env.render('rgb_array', width=240, height=240)
plt.imshow(img)
```

    <matplotlib.image.AxesImage at 0x122a83470>

[image]