google-deepmind/dm_control #128: Rendering Issue on Windows 10 (x64): GLFWError: (65539)

thegamejammer, 2020-01-10:
I know windows isn't officially supported on windows, but I managed to get dm_control working last summer and am since trying to continue that work with a new install.

dm_control and MuJoCo appears to mostly work, I can run simulations and use most functions. The only issue is that trying to render with `pixels = env.physics.render()` results in the following error:

```
runfile('C:/Users/user/OneDrive/Documents/FYP/python/DM Control Rendering.py', wdir='C:/Users/user/OneDrive/Documents/FYP/python')
Traceback (most recent call last):

  File "<ipython-input-5-346c1f7f484f>", line 1, in <module>
    runfile('C:/Users/user/OneDrive/Documents/FYP/python/DM Control Rendering.py', wdir='C:/Users/user/OneDrive/Documents/FYP/python')

  File "C:\Users\user\Anaconda3\lib\site-packages\spyder_kernels\customize\spydercustomize.py", line 827, in runfile
    execfile(filename, namespace)

  File "C:\Users\user\Anaconda3\lib\site-packages\spyder_kernels\customize\spydercustomize.py", line 110, in execfile
    exec(compile(f.read(), filename, 'exec'), namespace)

  File "C:/Users/user/OneDrive/Documents/FYP/python/DM Control Rendering.py", line 17, in <module>
    pixels = env.physics.render()

  File "C:\Users\user\Anaconda3\lib\site-packages\dm_control\mujoco\engine.py", line 183, in render
    physics=self, height=height, width=width, camera_id=camera_id)

  File "C:\Users\user\Anaconda3\lib\site-packages\dm_control\mujoco\engine.py", line 602, in __init__
    if self._physics.contexts.mujoco is not None:

  File "C:\Users\user\Anaconda3\lib\site-packages\dm_control\mujoco\engine.py", line 459, in contexts
    self._make_rendering_contexts()

  File "C:\Users\user\Anaconda3\lib\site-packages\dm_control\mujoco\engine.py", line 443, in _make_rendering_contexts
    max_width=max_width, max_height=max_height)

  File "C:\Users\user\Anaconda3\lib\site-packages\dm_control\_render\glfw_renderer.py", line 49, in __init__
    executor.PassthroughRenderExecutor)

  File "C:\Users\user\Anaconda3\lib\site-packages\dm_control\_render\base.py", line 68, in __init__
    ctx.call(self._platform_init, max_width, max_height)

  File "C:\Users\user\Anaconda3\lib\site-packages\dm_control\_render\executor\render_executor.py", line 142, in call
    return func(*args, **kwargs)

  File "C:\Users\user\Anaconda3\lib\site-packages\dm_control\_render\glfw_renderer.py", line 59, in _platform_init
    glfw.window_hint(glfw.DOUBLEBUFFER, 0)

  File "C:\Users\user\Anaconda3\lib\site-packages\glfw\__init__.py", line 1110, in window_hint
    _glfw.glfwWindowHint(hint, value)

  File "C:\Users\user\Anaconda3\lib\site-packages\glfw\__init__.py", line 609, in errcheck
    _reraise(exc[1], exc[2])

  File "C:\Users\user\Anaconda3\lib\site-packages\glfw\__init__.py", line 45, in _reraise
    raise exception.with_traceback(traceback)

  File "C:\Users\user\Anaconda3\lib\site-packages\glfw\__init__.py", line 588, in callback_wrapper
    return func(*args, **kwargs)

  File "C:\Users\user\Anaconda3\lib\site-packages\glfw\__init__.py", line 800, in _raise_glfw_errors_as_exceptions
    raise GLFWError(message)

GLFWError: (65539) b'Invalid argument for enum parameter'
```

While reading through other issues here on github, I have confirmed that `import glfw` and `assert glfw.init()` both work.

Any ideas? Happy to give more info. Many thanks for reading and helping.

alimuldal, 2020-01-13:
GLFW is complaining that `glfw.DOUBLEBUFFER` is an invalid enum parameter for `glfw.window_hint`. My guess its that it's not supported for Windows - according to [the GLFW docs](https://www.glfw.org/docs/latest/window_guide.html#window_hints_x11) it seems to be X11-specific. Perhaps try commenting out [this line](https://github.com/deepmind/dm_control/blob/master/dm_control/_render/glfw_renderer.py#L59)?

thegamejammer, 2020-01-13:
Hello, thank you for the response. I tried that and I got another error somewhere else instead (but still when I tried to render). I managed to fix the problem by taking a fresh `mujoco200 win64` download from https://www.roboti.us/index.html. I then took the glfw3.dll from the unzipped download and replaced the glfw3.dll at C:\Users\<user>\Anaconda3\Lib\site-packages\glfw with the windows64 specific .dll. This appeared to fix all problems and I got the render to work. I'm not exactly sure what was wrong with the old library, but I kept a copy of the old .dll and the working .dll for archival purposes.

closed 2020-01-13 by thegamejammer