google-deepmind/dm_control #129: Headless rendering on ubuntu machine

xinleipan, 2020-01-10:
I'm trying to get visual observations from a headless Ubuntu machine with `MUJOCO_GL=egl`, and got the following error report:

        AttributeError: 'GLXPlatform' object has no attribute 'EGL'

The same code can run on a desktop ubuntu with monitor without error report. It seems like this is related with the pyopengl package but I'm not sure how to solve this.

xinleipan, 2020-01-11:
Solved. Solution: install PyOpenGL 3.1.4 and PyOpenGL_accelerate 3.1.4 and export MUJOCO_GL=osmesa and export PYOPENGL_PLATFORM=osmesa

alimuldal, 2020-01-13:
You shouldn't have to `export PYOPENGL_PLATFORM` - this environment variable gets set automatically when you import either the OSMesa or EGL rendering backends (e.g. see [here](https://github.com/deepmind/dm_control/blob/master/dm_control/_render/pyopengl/osmesa_renderer.py#L29-L30)). One possible reason why you might encounter that `AttributeError` is if `PYOPENGL_PLATFORM` and `MUJOCO_GL` are set to mismatching values.

alimuldal, 2020-01-13:
You shouldn't need to set `PYOPENGL_PLATFORM` explicitly - this environment variable gets set automatically when either of the OSMesa or EGL rendering backends is imported (e.g. see [here](https://github.com/deepmind/dm_control/blob/master/dm_control/_render/pyopengl/osmesa_renderer.py#L27-L30)).

alimuldal, 2020-01-13:
Are you importing modules from `OpenGL` elsewhere in your code by any chance? That error message suggests that `GLXPlatform` (which is the default on Linux) has already been imported. My guess is that somewhere in your code you're importing from `OpenGL` before we've had a chance to set `PYOPENGL_PLATFORM=egl` inside `egl_renderer.py`.

xinleipan, 2020-01-23:
Thanks. I actually somehow solved this problem and it didn't appear anymore.

closed 2020-01-24 by alimuldal

wookayin, 2020-07-15:
For anyone who might run into this issue and find this issue from Googling: make sure you have `libegl1` (or `libegl-dev`) packages installed.

wsywsywsywsywsy979, 2022-11-02:
> Solved. Solution: install PyOpenGL 3.1.4 and PyOpenGL_accelerate 3.1.4 and export MUJOCO_GL=osmesa and export PYOPENGL_PLATFORM=osmesa
hi , if my code must use pyopengl3.1.0，and I have added "os.environ["PYOPENGL_PLATFORM"] = "osmesa"" before "import pyrender" ，but I get an error that “'GLXPlatform' object has no attribute 'OSMesa'”?could you give me some advice to solve this problem?

wsywsywsywsywsy979, 2022-11-02:
> 

hi ， I have tried "os.environ['PYOPENGL_PLATFORM'] = 'egl' ", but get an error "GLError(
        err = 12297,
        baseOperation = eglCreateContext,
        cArguments = (
                <OpenGL._opaque.EGLDisplay_pointer object at 0x7f77438fc510>,
                <OpenGL._opaque.EGLConfig_pointer object at 0x7f77438fc400>,
                <OpenGL._opaque.EGLContext_pointer object at 0x7f7743b5a1e0>,
                <OpenGL.arrays.lists.c_int_Array_7 object at 0x7f77438fc950>,
        ),
        result = <OpenGL._opaque.EGLContext_pointer object at 0x7f77438fce18>
)"， could you give me more advice to solve this problem?

IridescentJiang, 2024-08-01:
> Are you importing modules from `OpenGL` elsewhere in your code by any chance? That error message suggests that `GLXPlatform` (which is the default on Linux) has already been imported. My guess is [...]

Thanks! `export PYOPENGL_PLATFORM=egl` works for me!