google-deepmind/dm_control #131: can we change the robot physical parameters?

xinleipan, 2020-01-23:
So I created environments using the physics module and change some parameters such as body_mass to another vector that is slightly perturbed from the original vector. Will the simulation results still be correct?

wpumacay, 2020-01-23:
Hi @xinleipan . Perhaps [this](http://www.mujoco.org/forum/index.php?threads/can-the-parameters-of-a-model-be-changed-on-the-fly.3354/#post-4963) post from the MuJoCo forum could help. For masses, it seems that you'd have to make a call to `mj_setConst(model,data)` to recompute some quantities based on the *zero configuration*. I haven't tried out myself with dm_control, but with the C-API works ok (at least changing geoms params like sizes, mesh and hfield data, etc.). 

The call with dm_control seems to be:

```python
from dm_control.mujoco.wrapper.mjbindings import mjlib
## ...
mjlib.mj_setConst(m, d)
```
For how to use `mjlib` (exposes the C-API), I think you could look at the [core.py](https://github.com/deepmind/dm_control/blob/eab27eb2be37a8fb75b7bc8be131bc4fda3ccee4/dm_control/mujoco/wrapper/core.py#L455) file. There they make some calls using mjlib, just grab the model and data from the Physics object (`self.model`, `self.data` fields of the Physics object).

xinleipan, 2020-01-23:
Thanks. So directly modify the physical parameters such as people usually do with env.model.body_mass won't work?

wpumacay, 2020-01-23:
I'm sorry, didn't know you could grab the `model` directly from `env` :sweat_smile: . Perhaps this is related #106 . As they mention, the mjData and mjModel C-structs are exposed, so changing the fields should suffice. Emo mentioned in the [post](http://www.mujoco.org/forum/index.php?threads/can-the-parameters-of-a-model-be-changed-on-the-fly.3354/#post-4963) that only for model parameters that require to recompute some internals that call to `setConst` might be required along with the change on parameters itself, like the call to `env.model.body_mass` . If setting `env.model.body_mass` doesn't work as you expected, you could try **add** the extra call to `mjlib.mj_setConst` (that might do the trick).

xinleipan, 2020-01-23:
Thanks. I have another question, so when i changed the body_mass, and reset the environment and use physics.set_state() to set the system at the same state, why is the rendered image not exactly the same?If the system state is the same, should the rendered images be exactly the same?

wpumacay, 2020-01-23:
Mmmm, could you provide a little more information regarding the issue?. Perhaps a screenshot of the simulation for both cases, and also a gist to run and reproduce the issue. My guess would be that you changed the parameter you wanted, but because the parameters have changed you'd end up with a slightly different behaviour during simulation, e.g. if you changed the mass of a link in the model, then due to stiffness and damping in the joints, the link might get into a different state. I think the state should be the same, as it's been tested in a unit-test [here](https://github.com/deepmind/dm_control/blob/eab27eb2be37a8fb75b7bc8be131bc4fda3ccee4/dm_control/mujoco/engine_test.py#L267).

Also, did you get to solve the previous issue?. If this is a separate issue perhaps it'd be better to close this one and open a new one. You could check if the parameters changed by saving the mjModel, using `mjlib.mj_saveLastXML` or [`core.save_last_parsed_model_to_xml`](https://github.com/deepmind/dm_control/blob/eab27eb2be37a8fb75b7bc8be131bc4fda3ccee4/dm_control/mujoco/wrapper/core.py#L363) (the latter seems to be the one exposed) and checking that the parameters you changed during runtime were changed in the XML. Here's a [test](https://github.com/deepmind/dm_control/blob/eab27eb2be37a8fb75b7bc8be131bc4fda3ccee4/dm_control/mujoco/wrapper/core_test.py#L131) where this function is used.

Out of curiosity, did you get to change the mass of a specific link you wanted? or are you setting the whole mass of the model with `env.model.body_mass` directly?. Perhaps what you want is the first option, so you'd have to find the **id** of the link based on its name in the model with `physics.model.name2id`, and use this id as an index in the whole vector `body_mass` to set the mass you wanted for that link in specific. You might be changing accidentally the masses of a lot of bodies in the simulation, which in the case of there being obstacles in the scene, you might have accidentally changed those as well. A way to avoid using the indices is described in #106, using `env.physics.named` instead.

xinleipan, 2020-01-23:
Thanks. I created a repo here: https://github.com/xinleipan/dm_control for reproducing the rendering bugs.

yuvaltassa, 2020-01-23:
The Cheetah has [dynamics-based initialisation](https://github.com/deepmind/dm_control/blob/eab27eb2be37a8fb75b7bc8be131bc4fda3ccee4/dm_control/suite/cheetah.py#L77), so changing dynamic parameters will affect the initial state.

closed 2020-01-23 by yuvaltassa

xinleipan, 2020-01-24:
Thanks. But I didn’t change the dynamics in the example I provided?

wpumacay, 2020-01-24:
Hello @xinleipan. I think the last comment could explain the issue when you change the dynamic parameters, but in your example, I think you haven't changed those. Perhaps the issue might be related to how the scene (for visualization) is updated from the current "state" of the simulation. Basically, the flow it's something like this (`state` here is `[qpos, qvel, qact]`):

```
                mj_step                mjv_updateScene                mjr_render
[state]---->[xpos,xvel,...]----> (update visualization state) -> (render actual frame)
```

In your case, I think you took a `env.step(random_action)`, so your whole internal structures are set to that full different configuration.

* You first reset the environment, and landed in state `s1`, and "full configuration" `f1`. You then took a frame of that full configuration.
* Then you took an action and laded in another state `s2`, and "full configuration" `f2`. You then changed the state to `s1` and took a frame there, but I'm pretty sure that the information used to render the frame is from the full configuration (xpos), which is `f2`, not `f1`, so you see the scene as it was before, which causes the mismatch.

Just add the following after line 143:

```python
mjlib.mj_step1(env.env._physics.model.ptr, env.env._physics.data.ptr)
```

This will make a step that doesn't require of actuation controls, which gets you from state `s1` to the full configuration for that state `f1`. I got the frames showing the same.

yuvaltassa, 2020-01-24:
Ah sorry, I think I misunderstood earlier. Yes, @wpumacay is indeed correct. In order to propagate all the state-dependent quantities, step1 needs to be called. The `dm_control` way of doing this is with `physics.teset_context()`. Please see the discussion that begins at the bottom of page 10 in our [tech report](https://arxiv.org/abs/1801.00690).

xinleipan, 2020-01-24:
Thanks. I tried again and by adding the physics.reset_context() it works now. By the way, I’m wondering if I want to change the dynamics of the environment and by changing the physics.model.body_mass for example, could I just directly change it and reset the environment?