google-deepmind/dm_control #132: Mujoco Soccer simulation too slow

Alanliaoyihang, 2020-01-25:
I am trying to train a 1v1 soccer games in the soccer environment. However, the training is really slow. Is there any method to speed up the simulation?

lich14, 2020-04-03:
Since you are running soccer. Do you have any idea on what each observation truly means.

Frank-Dz, 2022-03-19:
> Since you are running soccer. Do you have any idea on what each observation truly means.

I have the same question. Refer to my post:
https://github.com/deepmind/dm_control/issues/271

Frank-Dz, 2022-03-19:
> I am trying to train a 1v1 soccer games in the soccer environment. However, the training is really slow. Is there any method to speed up the simulation?

I also met the same problem. 
The slowest part seems like the reset process after each episode.