google-deepmind/dm_control #133: No module named LabMaze in labmaze_texture

domhuh, 2020-02-26:
Traceback (most recent call last):
  File "rodent.py", line 1, in <module>
    from dm_control.locomotion.arenas import corridors as corr_arenas
  File "/Library/Frameworks/Python.framework/Versions/3.7/lib/python3.7/site-packages/dm_control/locomotion/arenas/__init__.py", line 22, in <module>
    from dm_control.locomotion.arenas.labmaze_textures import FloorTextures
  File "/Library/Frameworks/Python.framework/Versions/3.7/lib/python3.7/site-packages/dm_control/locomotion/arenas/labmaze_textures.py", line 23, in <module>
    from labmaze import assets as labmaze_assets
ModuleNotFoundError: No module named 'labmaze'

saran-t, 2020-02-26:
The maze generator is listed as an extra requirement for dm_control. You could either:
- Install `dm_control` with `locomotion_mazes` feature: `pip install dm_control[locomotion_mazes]`, or
- Manually install `labmaze`: `pip install labmaze`

I'll update our documentation as well. Thanks for bringing it up.

saran-t, 2020-09-17:
LabMaze is now listed as a dependency for dm_control generally so this issue is now obsolete.

closed 2020-09-17 by saran-t