google-deepmind/dm_control #135: Cannot install with Python3.8

PiggyGenius, 2020-03-02:
After  unzipping mujoco in ~/.mujoco/ and installing bazel, pip install dm_control gives the following error:
`I
nstalling collected packages: dm-tree, dm-env, dm-control
    Running setup.py install for dm-tree ... error
    ERROR: Command errored out with exit status 1:
     command: /home/ludovic/.general_env/bin/python -u -c 'import sys, setuptools, tokenize; sys.argv[0] = '"'"'/tmp/pip-install-21tka1qj/dm-tree/setup.py'"'"'; __file__='"'"'/tmp/pip-install-21tka1qj/dm-tree/setup.py'"'"';f=getattr(tokenize, '"'"'open'"'"', open)(__file__);code=f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();exec(compile(code, __file__, '"'"'exec'"'"'))' install --record /tmp/pip-record-ou4ug0ty/install-record.txt --single-version-externally-managed --compile --install-headers /home/ludovic/.general_env/include/site/python3.8/dm-tree
         cwd: /tmp/pip-install-21tka1qj/dm-tree/
    Complete output (37 lines):
    running install
    running build
    running build_py
    creating build
    creating build/lib.linux-x86_64-3.8
    creating build/lib.linux-x86_64-3.8/tree
    copying tree/tree_test.py -> build/lib.linux-x86_64-3.8/tree
    copying tree/tree_benchmark.py -> build/lib.linux-x86_64-3.8/tree
    copying tree/__init__.py -> build/lib.linux-x86_64-3.8/tree
    running build_ext
    bazel build //tree:_tree --symlink_prefix=build/temp.linux-x86_64-3.8/bazel- --compilation_mode=opt
    Starting local Bazel server and connecting to it...
    Loading:
    Loading: 0 packages loaded
    Analyzing: target //tree:_tree (1 packages loaded, 0 targets configured)
    INFO: Analyzed target //tree:_tree (23 packages loaded, 286 targets configured).
    INFO: Found 1 target...
    [0 / 26] [Prepa] BazelWorkspaceStatusAction stable-status.txt
    ERROR: /tmp/pip-install-21tka1qj/dm-tree/tree/BUILD:27:1: C++ compilation of rule '//tree:_tree.so' failed (Exit 1) gcc failed: error executing command /usr/bin/gcc -U_FORTIFY_SOURCE -fstack-protector -Wall -Wunused-but-set-parameter -Wno-free-nonheap-object -fno-omit-frame-pointer -g0 -O2 '-D_FORTIFY_SOURCE=1' -DNDEBUG -ffunction-sections ... (remaining 47 argument(s) skipped)
    
    Use --sandbox_debug to see verbose messages from the sandbox
    In file included from external/python_headers/object.h:4,
                     from external/python_headers/pytime.h:6,
                     from external/python_headers/Python.h:85,
                     from ./tree/tree.h:19,
                     from tree/tree.cc:15:
    external/python_headers/pymem.h:142:13: fatal error: cpython/pymem.h: No such file or directory
      142 | #  include  "cpython/pymem.h"
          |             ^~~~~~~~~~~~~~~~~
    compilation terminated.
    Target //tree:_tree failed to build
    Use --verbose_failures to see the command lines of failed build steps.
    INFO: Elapsed time: 2.571s, Critical Path: 0.21s
    INFO: 1 process: 1 linux-sandbox.
    FAILED: Build did NOT complete successfully
    FAILED: Build did NOT complete successfully
    error: command 'bazel' failed with exit status 1
    ----------------------------------------
ERROR: Command errored out with exit status 1: /home/ludovic/.general_env/bin/python -u -c 'import sys, setuptools, tokenize; sys.argv[0] = '"'"'/tmp/pip-install-21tka1qj/dm-tree/setup.py'"'"'; __file__='"'"'/tmp/pip-install-21tka1qj/dm-tree/setup.py'"'"';f=getattr(tokenize, '"'"'open'"'"', open)(__file__);code=f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();exec(compile(code, __file__, '"'"'exec'"'"'))' install --record /tmp/pip-record-ou4ug0ty/install-record.txt --single-version-externally-managed --compile --install-headers /home/ludovic/.general_env/include/site/python3.8/dm-tree Check the logs for full command output.
`

I am attempting to install with python3.8, I tried the same procedure with python3.6 and had no issue. I played around in the install files and I assume the problem comes from the installer not being able to find the relative include: cpython/<HEADER>.h (they do exist despite the No such file error).

In the example above, the problem comes from cpython/pymem.h but removing the dependency in the pymem.h file just moves the problem to another header file that depends on a file in cpython/

alimuldal, 2020-03-03:
`dm-tree` doesn't officially support Python 3.8 and there are no pre-built wheels for that particular version. However I'm able to build it from source without problems.

The error message suggests that you're missing the header files for Python 3.8 - could you try installing them (`apt-get install python3.8-dev` on Ubuntu)?

PiggyGenius, 2020-03-03:
I will try this on Ubuntu (I am building python from sources so I might be missing some stuff). But if that was the (only) problem, the install should work on archlinux (since there is no dev package).

alimuldal, 2020-03-13:
Did you try building from source on Ubuntu with the `python3.8-dev` package installed?

PiggyGenius, 2020-03-17:
I didn't get the chance to try that on ubuntu, I changed my docker to use python 3.7. I still have the problem on archlinux though.

KeAWang, 2020-03-26:
I'm having the same problem on python 3.7 in a conda environment with archlinux

alimuldal, 2020-04-18:
I was able to install `dm_control` in a Python 3.8 pipenv within an Archlinux VM.
```
(dm_control_vm) [vagrant@archlinux dm_control_vm]$ uname -a
Linux archlinux 5.5.13-arch2-1 #1 SMP PREEMPT Mon, 30 Mar 2020 20:42:41 +0000 x86_64 GNU/Linux
(dm_control_vm) [vagrant@archlinux dm_control_vm]$ python --version
Python 3.8.2
(dm_control_vm) [vagrant@archlinux dm_control_vm]$ bazel --version
bazel 2.1.0- (@non-git)
(dm_control_vm) [vagrant@archlinux dm_control_vm]$ cc --version
cc (Arch Linux 9.2.1+20200130-2) 9.2.1 20200130
Copyright (C) 2019 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warr
(dm_control_vm) [vagrant@archlinux dm_control_vm]$ pipenv graph
dm-control==0.0.300771433
  - absl-py [required: >=0.7.0, installed: 0.9.0]
    - six [required: Any, installed: 1.14.0]
  - dm-env [required: Any, installed: 1.2]
    - absl-py [required: Any, installed: 0.9.0]
      - six [required: Any, installed: 1.14.0]
    - dm-tree [required: Any, installed: 0.1.4]
      - six [required: >=1.12.0, installed: 1.14.0]
    - numpy [required: Any, installed: 1.18.2]
    - six [required: Any, installed: 1.14.0]
  - dm-tree [required: !=0.1.2, installed: 0.1.4]
    - six [required: >=1.12.0, installed: 1.14.0]
  - future [required: Any, installed: 0.18.2]
  - glfw [required: Any, installed: 1.11.0]
  - lxml [required: Any, installed: 4.5.0]
  - numpy [required: Any, installed: 1.18.2]
  - pyopengl [required: >=3.1.4, installed: 3.1.5]
  - pyparsing [required: Any, installed: 2.4.7]
  - scipy [required: Any, installed: 1.4.1]
    - numpy [required: >=1.13.3, installed: 1.18.2]
  - setuptools [required: Any, installed: 46.1.3]
  - six [required: Any, installed: 1.14.0]
```

alimuldal, 2020-04-20:
@KeAWang It's not clear to me why you would encounter this problem with Python 3.7, since `dm-tree` has precompiled Python 3.7 wheels for Linux, OSX and Windows. Do you know why is it being compiled from source in your case?

alimuldal, 2020-04-20:
@KeAWang Have you tried doing `pip install dm-tree` within your conda environment? That way you'll get the precompiled wheel from PyPI, rather than having to compile it from source.

KeAWang, 2020-04-20:
Sorry I think I made a mistake. My env was actually 3.8 instead of 3.7. It works with 3.7. Thank you so much!

PiggyGenius, 2020-04-24:
@alimuldal I have tried again after updating bazel. I was able to install dm_control with no issues on python3.8 on a native archlinux system. Might have been an incoherent system error. Thank you very much for the feedback.

alimuldal, 2020-04-24:
OK, great - I'll close this then. Hopefully this will become easier once `dm-tree` release wheels for Python 3.8, but there's not much more we can do about it from our end.

closed 2020-04-24 by alimuldal

yash31415, 2021-03-15:
Please update your pip 
sudo pip3.8 install --upgrade pip