google-deepmind/dm_control #136: Headless rendering error when executing in Docker

vittorione94, 2020-03-09:
I'm having troubles executing a script in Docker 

The script is:
```
import os
os.environ['DISABLE_MUJOCO_RENDERING'] = '1'

from dm_control import suite

env = suite.load(domain_name="humanoid", task_name="stand")
action_spec = env.action_spec()

print(action_spec)

```

What doesn't make sense to me is that when I ssh the machine and execute this script it works without a problem but after building a docker image and running it I get the following:
```
Traceback (most recent call last):
  File "test_dm_control.py", line 4, in <module>
    from dm_control import suite
  File "/usr/local/lib/python3.6/dist-packages/dm_control/suite/__init__.py", line 28, in <module>
    from dm_control.suite import acrobot
  File "/usr/local/lib/python3.6/dist-packages/dm_control/suite/acrobot.py", line 24, in <module>
    from dm_control import mujoco
  File "/usr/local/lib/python3.6/dist-packages/dm_control/mujoco/__init__.py", line 18, in <module>
    from dm_control.mujoco.engine import action_spec
  File "/usr/local/lib/python3.6/dist-packages/dm_control/mujoco/engine.py", line 44, in <module>
    from dm_control import _render
  File "/usr/local/lib/python3.6/dist-packages/dm_control/_render/__init__.py", line 75, in <module>
    Renderer = import_func()
  File "/usr/local/lib/python3.6/dist-packages/dm_control/_render/__init__.py", line 36, in _import_egl
    from dm_control._render.pyopengl.egl_renderer import EGLContext
  File "/usr/local/lib/python3.6/dist-packages/dm_control/_render/pyopengl/egl_renderer.py", line 43, in <module>
    from dm_control._render.pyopengl import egl_ext as EGL
  File "/usr/local/lib/python3.6/dist-packages/dm_control/_render/pyopengl/egl_ext.py", line 33, in <module>
    from OpenGL import EGL
  File "/usr/local/lib/python3.6/dist-packages/OpenGL/EGL/__init__.py", line 2, in <module>
    from OpenGL.raw.EGL._types import *
  File "/usr/local/lib/python3.6/dist-packages/OpenGL/raw/EGL/_types.py", line 87, in <module>
    raw_eglQueryString = _p.PLATFORM.EGL.eglQueryString
AttributeError: 'NoneType' object has no attribute 'eglQueryString'
```
The docker file is:
```
# Use the official image as a parent image
FROM tensorflow/tensorflow:2.0.0-gpu-py3

ENV MJLIB_PATH=/tf/mujoco200_linux/bin/libmujoco200.so
ENV MJKEY_PATH=/tf/mujoco200_linux/bin/mjkey.txt
ENV LD_LIBRARY_PATH=/tf/mujoco200_linux/bin/
ENV PYGLFW_LIBRARY=/tf/mujoco200_linux/bin/libglfw.so.3

# i need this for ffmpeg
ENV PATH="/tf/.local/bin/:${PATH}"

# Run the command inside your image filesystem
RUN pip install --upgrade pip && \
        pip install gym && \
    pip install gym[atari] && \
        pip install dm_control && \
        pip install tqdm && \
        pip install np_utils && \
        pip install opencv-python
```
The OS both on the server and in Docker is Ubuntu.
Thanks a lot for any help you can provide!

alimuldal, 2020-03-10:
`DISABLE_MUJOCO_RENDERING` will have no effect since that environment variable was removed in 2f9f30051d8f898222db28a00953bf62043c519b. There's no longer a way to explicitly disable rendering - what _should_ happen now is that if none of the rendering backends can be imported then you'll get a deferred error if you later try to create a rendering context (see [here](https://github.com/deepmind/dm_control/blob/master/dm_control/_render/__init__.py#L82)). You can also force a particular choice of rendering backend by setting the `MUJOCO_GL=` environment variable (see the docs [here](https://github.com/deepmind/dm_control#rendering)).

What version of PyOpenGL are you using? Can you try running with `MUJOCO_GL=osmesa` instead?

vittorione94, 2020-03-10:
Thanks a lot for your reply! 

The version of PyOpenGL installed is 3.1.5

Even if I set `MUJOCO_GL=osmesa` I have the exact same error

vittorione94, 2020-03-10:
I don't know if this help but If I set `MUJOCO_GL=glfw` I have (I think correctly) the following error:

```
Traceback (most recent call last):
  File "/usr/local/lib/python3.6/dist-packages/dm_control/_render/glfw_renderer.py", line 30, in <module>
    import glfw  # pylint: disable=g-import-not-at-top
  File "/usr/local/lib/python3.6/dist-packages/glfw/__init__.py", line 41, in <module>
    raise ImportError("Failed to load GLFW3 shared library.")
ImportError: Failed to load GLFW3 shared library.

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "test_dm_control.py", line 4, in <module>
    from dm_control import suite
  File "/usr/local/lib/python3.6/dist-packages/dm_control/suite/__init__.py", line 28, in <module>
    from dm_control.suite import acrobot
  File "/usr/local/lib/python3.6/dist-packages/dm_control/suite/acrobot.py", line 24, in <module>
    from dm_control import mujoco
  File "/usr/local/lib/python3.6/dist-packages/dm_control/mujoco/__init__.py", line 18, in <module>
    from dm_control.mujoco.engine import action_spec
  File "/usr/local/lib/python3.6/dist-packages/dm_control/mujoco/engine.py", line 44, in <module>
    from dm_control import _render
  File "/usr/local/lib/python3.6/dist-packages/dm_control/_render/__init__.py", line 67, in <module>
    Renderer = import_func()  # pylint: disable=invalid-name
  File "/usr/local/lib/python3.6/dist-packages/dm_control/_render/__init__.py", line 41, in _import_glfw
    from dm_control._render.glfw_renderer import GLFWContext
  File "/usr/local/lib/python3.6/dist-packages/dm_control/_render/glfw_renderer.py", line 33, in <module>
    six.reraise(ImportError, ImportError(str(exc)), tb)
  File "/usr/lib/python3/dist-packages/six.py", line 692, in reraise
    raise value.with_traceback(tb)
  File "/usr/local/lib/python3.6/dist-packages/dm_control/_render/glfw_renderer.py", line 30, in <module>
    import glfw  # pylint: disable=g-import-not-at-top
  File "/usr/local/lib/python3.6/dist-packages/glfw/__init__.py", line 41, in <module>
    raise ImportError("Failed to load GLFW3 shared library.")
ImportError: Failed to load GLFW3 shared library.
```

vittorione94, 2020-03-10:
> Thanks a lot for your reply!
> 
> The version of PyOpenGL installed is 3.1.5
> 
> Even if I set `MUJOCO_GL=osmesa` I have the following error
> 
> ```
> Traceback (most recent call last):
>   File "test_dm_control.py", line 4, in <module>
>     from dm_control import suite
>   File "/usr/local/lib/python3.6/dist-packages/dm_control/suite/__init__.py", line 28, in <module>
>     from dm_control.suite import acrobot
>   File "/usr/local/lib/python3.6/dist-packages/dm_control/suite/acrobot.py", line 24, in <module>
>     from dm_control import mujoco
>   File "/usr/local/lib/python3.6/dist-packages/dm_control/mujoco/__init__.py", line 18, in <module>
>     from dm_control.mujoco.engine import action_spec
>   File "/usr/local/lib/python3.6/dist-packages/dm_control/mujoco/engine.py", line 44, in <module>
>     from dm_control import _render
>   File "/usr/local/lib/python3.6/dist-packages/dm_control/_render/__init__.py", line 67, in <module>
>     Renderer = import_func()  # pylint: disable=invalid-name
>   File "/usr/local/lib/python3.6/dist-packages/dm_control/_render/__init__.py", line 46, in _import_osmesa
>     from dm_control._render.pyopengl.osmesa_renderer import OSMesaContext
>   File "/usr/local/lib/python3.6/dist-packages/dm_control/_render/pyopengl/osmesa_renderer.py", line 39, in <module>
>     from OpenGL import GL
>   File "/usr/local/lib/python3.6/dist-packages/OpenGL/GL/__init__.py", line 4, in <module>
>     from OpenGL.GL.VERSION.GL_1_1 import *
>   File "/usr/local/lib/python3.6/dist-packages/OpenGL/GL/VERSION/GL_1_1.py", line 14, in <module>
>     from OpenGL.raw.GL.VERSION.GL_1_1 import *
>   File "/usr/local/lib/python3.6/dist-packages/OpenGL/raw/GL/VERSION/GL_1_1.py", line 7, in <module>
>     from OpenGL.raw.GL import _errors
>   File "/usr/local/lib/python3.6/dist-packages/OpenGL/raw/GL/_errors.py", line 4, in <module>
>     _error_checker = _ErrorChecker( _p, _p.GL.glGetError )
> AttributeError: 'NoneType' object has no attribute 'glGetError'
> ```

Googling around seems this is a common issue:

https://github.com/mcfletch/pyopengl/issues/10
https://github.com/NixOS/nixpkgs/issues/76822
https://github.com/MPI-IS/mesh/issues/23
https://sourceforge.net/p/pyopengl/mailman/message/35237182/

vittorione94, 2020-03-10:
I solved the issue with the following Dockerfile:

```
# Use the official image as a parent image
FROM tensorflow/tensorflow:2.0.0-gpu-py3

ENV MJLIB_PATH=/tf/mujoco200_linux/bin/libmujoco200.so
ENV MJKEY_PATH=/tf/mujoco200_linux/bin/mjkey.txt
ENV MUJOCO_GL=osmesa
ENV LD_LIBRARY_PATH=/tf/mujoco200_linux/bin/
ENV PYOPENGL_PLATFORM=osmesa

# i need this for ffmpeg
ENV PATH="/tf/.local/bin/:${PATH}"

RUN apt-cache search mesa
RUN apt-get update && apt-get install -y libgl1-mesa-glx libosmesa6

# Run the command inside your image filesystem
RUN pip install --upgrade pip && \
        pip install gym && \
        pip install gym[atari] && \
        pip install PyOpenGL==3.1.0 && \
        pip install dm_control && \
        pip install tqdm && \
        pip install np_utils && \
        pip install opencv-python
```

closed 2020-03-10 by vittorione94

alimuldal, 2020-03-11:
Ah I see, you were missing the `libgl1-mesa-glx` and `libosmesa6` deps described [here](https://github.com/deepmind/dm_control#rendering)

gabriel-hurtado, 2021-05-04:
I solved a similar bug by doing `pip install pyrender`

rxng8, 2023-09-16:
* Do this if you have EGL error
  * Install mesalib first to use osmesa render instead of egl: `conda install -c conda-forge mesalib`
  * Then change the environment variable to osmesa:
  ```
  import os
  os.environ["PYOPENGL_PLATFORM"] = "osmesa"
  os.environ["MUJOCO_GL"] = "osmesa"
  ```

RylonW, 2023-12-22:
> * Do this if you have EGL error
>   
>   * Install mesalib first to use osmesa render instead of egl: `conda install -c conda-forge mesalib`
>   * Then change the environment variable to osmesa:
>   
>   ```
>   import os
>   os.environ["PYOPENGL_PLATFORM"] = "osmesa"
>   os.environ["MUJOCO_GL"] = "osmesa"
>   ```

Thanks to rxng8. The reply works for me. My code is shown below:
```
import mujoco
mujoco.MjModel.from_xml_string('<mujoco/>')
```
It runs successfully and outputs:
```
<mujoco._structs.MjModel at 0x7f5ad8c13d70>
```

amoskalev, 2025-09-19:
> * Do this if you have EGL error
>   
>   * Install mesalib first to use osmesa render instead of egl: `conda install -c conda-forge mesalib`
>   * Then change the environment variable to osmesa:
>   
>   ```
>   import os
>   os.environ["PYOPENGL_PLATFORM"] = "osmesa"
>   os.environ["MUJOCO_GL"] = "osmesa"
>   ```

this solves it for me