google-deepmind/dm_control #137: How can I get the true meaning of states

lich14, 2020-03-16:
I have read corresponding paper of this environment. But I am wondering what is the true meaning of each feature. Such as the Cartpole environment, there are four positions and two speeds. I wish to know what is it.

wpumacay, 2020-03-17:
Hi @lich14 . You could take a look at the [cartpole.py](https://github.com/deepmind/dm_control/blob/d677e2b57f20a85211e4cedd4b02012b254ab023/dm_control/suite/cartpole.py) file, which has the implementation of that environment. From the [get_observation](https://github.com/deepmind/dm_control/blob/d677e2b57f20a85211e4cedd4b02012b254ab023/dm_control/suite/cartpole.py#L202) method it seems that it returns the *"state"* given by the **bounded_position** and **velocity** methods. I think the **velocity** method returns the pair of velocities of the two degrees of freedom of your system (1 slide-joint <> cart position, and 1 hinge-joint <> pole angle). 

The **bounded_position** method seems to return the cart_position (which is just 1 value, from qpos[0]), and the last row of the rotation matrix of the pole (given by the xmat[2:,['zz','xz']]. It looks from the signature that it should return only the *zz* and *xz* from the last row, but perhaps I'm mistaken. For further reference regarding the elements of the model struct, you could take a look at [this](http://mujoco.org/book/APIreference.html#mjData) reference from the mujoco-API documentation (look for **xmat**). dm_control exposes bindings of those structs via ctypes, so you can directly map those structs (mainly **mjMode**l and **mjData**) to the exposed bindings.

alimuldal, 2020-03-17:
What @wpumacay said is correct. `xmat[2:,['zz','xz']]` corresponds to the sine and cosine of the pole angle. These are more convenient as inputs to neural networks than the raw angle because they are bounded between -1 and 1, and there are no discontinuities when the pole spins > 360° in either direction.

closed 2020-03-17 by alimuldal