google-deepmind/dm_control #139: how can I sample a random state from an environment?

tungnd1705, 2020-03-29:
In my project, I need to be able to sample a random but valid state (the underlying physics state, not the observation) from any environment, let's say cartpole.
In order to do that, I have to know the range of each element in qpos and qvel, however I cannot find any of those information online.
Can anyone help me with this please?

wpumacay, 2020-03-30:
Hi @tungnd1705 . I think you'll have to access the internals of the mujoco-model structure (mjModel) in order to grab the information for the ranges of the joints (if these have a range). E.g. for cartpole there're two joints (see [cartpole.xml](https://github.com/deepmind/dm_control/blob/master/dm_control/suite/cartpole.xml)):

* **The slide joint for the cart** has a range of -1.8, +1.8, which you can grab from mjModel->jnt_range[ID_OF_SLIDE_JOINT], which is an array of 2 floats (min-max). You can access these fields by using the python bindings. I think the call could be something like this (the `physics` structure can be obtained from `env.env._physics`, or some other access point in the environment wrapper):
```python
# get the id of the slide-joint (joint-name="slider")
jnt_id = physics.model.name2id['slider']
# grab the range of the slide joint (should be an ndarray of length 2)
jnt_range = physics.model.jnt_range[jnt_id]
```
* **The hinge joint for the cart** has no range. Because it's an angle, you could assume it goes from `[0-2*np.pi]`. If this joint had a range, then you could access it in a similar way as the above example.
-----
**Note**: *I'm not quite sure regarding the bindings, but I think the core developers included support to access geoms, bodies and joints by their names, which would be a better option over having to grab the `id` and then grabbing the required information. Check [this](https://github.com/deepmind/dm_control/blob/03bebdf9eea0cbab480aa4882adbc4184b850835/dm_control/suite/cartpole.py#L138) method, in which they access a field called `named` for name-based access*

**Note**: *I'm not sure if there are ways to access min-max ranges for qvel, as they can be in principle "unbounded" (unless simulation becomes unstable). I would use some reasonable defaults for these though, like some not-to-large ranges for velocities, etc.*

In order to avoid having to check your model for each joint, you could iterate all valid joints of your model and check if they have range or not (querying the `model.jnt_limited` field). If you have limits, you grab them as in the example above, and if not, you could set them according to the joint-type, which you could grab using `model.jnt_type`, and if they're hinge-joints, you set a default of `[0-2*np.pi]`, if they're slide-joints you could set a default of `[-np.inf, np.inf]`, etc. I think the joints are parsed in order, so if you iterate over `model.njnt` (number of joints), then they relate in increasing order with their respective qpos. Also, the only way to increase add a qpos is by adding a joint (which can add 1,3,4,7 qpos values, depending of the joint-type), so I'm pretty sure all your q-values for the random-state you want to sample would have the correct ranges (if any).

Here's the Mujoco API-reference in case you need it [api-ref.](http://mujoco.org/book/APIreference.html#mjModel). If you still have any issues just post a comment :smile: .

tungnd1705, 2020-04-03:
thank you @wpumacay , you comment helped a lot
I also found [this](https://github.com/deepmind/dm_control/blob/03bebdf9eea0cbab480aa4882adbc4184b850835/dm_control/suite/utils/randomizers.py#L39), which is pretty all I need to sample a random state

wpumacay, 2020-04-03:
Great @tungnd1705 :smile: . I think you could close this issue if there's no more help needed related to this issue :+1: .

closed 2020-04-04 by tungnd1705