google-deepmind/dm_control #140: Cannot open STL file

zemus-lab, 2020-04-08:
Hi, does anyone know how I can solve this issue?

dm_control.mujoco.wrapper.core.Error: Error: could not open STL file

wpumacay, 2020-04-08:
Hi @zemus-lab . Is your .stl file in **stl binary format**?. I had some similar issues using mujoco C-API, and the error was related to the file format. You could try **opening it with mesh-lab**, and exporting it as **.stl binary** (there's a checkbox in the export options).

The related documentation for mesh assets can be found [here](http://www.mujoco.org/book/XMLreference.html#mesh):
*"MuJoCo works with triangulated meshes. They can be loaded from binary STL files, binary MSH files with custom format described below, or ..."*

Is your asset is in a public repo?, perhaps I could take a look at it locally.

zemus-lab, 2020-04-09:
Hi @wpumacay. Yes, it is in binary format. I loaded it without problem with simulate and basic from mujoco

wpumacay, 2020-04-09:
Mmm, that's weird. Could you provide the script you're trying to run? (Perhaps as a github-gist) to try to loaded locally, or check for any other errors.

zemus-lab, 2020-04-12:
What I did was taking the existing manipulator suite from dm and change the existing body of the manipulator for my model. I copied all the files here:  https://li_dl@bitbucket.org/li_dl/repo.git

wpumacay, 2020-04-19:
Hi @zemus-lab , sorry for the late response. I got to try your repo and could reproduce the error. Your model had a few issues, which I list below. Basically the issues were due to naming and folder/location:

* The names were in **some cases uppercase and some lowercase** (e.g. LINK3.stl, LINK4.stl). I renamed your .stl models in the folder to all lowercase. Also, the extension of the files was uppercase in the mjcf-model, but lowercase for the actual resources:

```xml
    <mesh name="base" file="stl_files/base.STL" scale="0.001 0.001 0.001"/>
```
Just update the extensions to all lowercase to fix this issue.
* The folder location from which the mujoco-compiler tries to load the meshes was pointing to the wrong place (not the wrong place, but your meshes folder wasn't being found by the compiler). To fix this issue just add the `meshdir` attribute to the `compiler` element in your mjcf-model, as follows:
```xml
<mujoco model="IRB120">
    <compiler angle="radian" coordinate="local" inertiafromgeom="true" settotalmass="25" meshdir="PATH-WHERE-THE-STL-FILES-FOLDER-IS"/>
   ...
```
Notice that your meshes in the model are located like **stl_files/base.stl**, so the `meshdir` should be the the folder where the folder called `stl_files` is located. In my case the meshes are in (base.stl for example) `/home/gregor/Documents/repos/dm_control/examples/repo/stl_files/base.stl`, so your meshdir should be `meshdir="/home/gregor/Documents/repos/dm_control/examples/repo"`

Using all previous fixes I got your model loaded and got the following in the visualizer:

[image]

Hope this was helpful. If you still have troubles just leave a comment :smile:. Otherwise, feel free to close this issue. (Btw, nice model :+1: )

zemus-lab, 2020-04-20:
It looks like it failed because I didn¡t add the meshdir part. 
Thanks!

closed 2020-04-20 by zemus-lab