google-deepmind/dm_control #141: Collision detection for multi-agent testing

CharlesAverill, 2020-04-09:
I'm writing a custom task for 2 agents in the same environment. I need to know when the 2 agents collide and which body parts of each are colliding for reward calculation. I'm imagining something similar to Unity's OnCollisionEnter. Is there a way to do this with dm_control?

wpumacay, 2020-04-09:
Hi @CharlesAverill . The contacts information is stored in the internal `mjData` structure, which is exposed via python bindings as `physics.data`. There's a field called `contacts` in this structure, which contains all the contacts ("collisions") found by the collision detector. Here's the [documentation](http://www.mujoco.org/book/APIreference.html#mjData) related to that structure. Look for the following part:
```c
    // ...
    // computed by mj_fwdPosition/mj_collision
    mjContact* contact;             // list of all detected contacts            (nconmax x 1)
    // ...
```
The number of detected contacts can be obtained via `physics.data.ncon`, which you can use to iterate over the `contact` field. Each element in that field is a `mjContact` structure, which is described [here](http://www.mujoco.org/book/APIreference.html#mjContact). I'm positive these are also exposed via the python bindings, so you can just grab the information you require for those contacts. If you're interested only on the geoms that collided, then just check for these fields:
```c
struct _mjContact                   // result of collision detection functions
{
    // ...
    int geom1;                      // id of geom 1
    int geom2;                      // id of geom 2
    // ...
};
```

If then you want to obtain the name of the geoms that collided (might be easier to check that way), you could use the function `mj_id2name`, which I think is exposed on the python package `mjbindings`

```python
from dm_control.mujoco.wrapper.mjbindings import mjlib
name = mjlib.mj_id2name( id_geom )
```

alimuldal, 2020-04-14:
`contact` is an `np.recarray` that exposes the struct fields as vectors, so you can vectorize the filtering code, which may help performance if you have lots of collisions:

```python
agent_1_geom_ids = ...
agent_2_geom_ids = ...

geom1 = env.physics.data.contact.geom1
geom2 = env.physics.data.contact.geom2
agent1to2 = np.in1d(geom1, agent1_geom_ids) & np.in1d(geom2, agent2_geom_ids)
agent2to1 = np.in1d(geom1, agent2_geom_ids) & np.in1d(geom2, agent1_geom_ids)
is_relevant = agent1to2 | agent2to1

for contact in env.physics.data.contact[is_relevant]:
  geom_names = [env.physics.model.id2name(g) for g in [contact.geom1, contact.geom2]]
  print(', '.join(geom_names))
```

closed 2020-04-18 by alimuldal