google-deepmind/dm_control #145: dm_control.mujoco.wrapper.core.Error:Could not register license

kashifzr, 2020-05-26:
- First I set the Environment Variables with the following command
export MUJOCO_GL="glfw"
export MJLIB_PATH=$HOME/.mujoco/mujoco200/bin/libmujoco200.dylib
export MJKEY_PATH=$HOME/.mujoco/mujoco200/mjkey.txt
export LD_LIBRARY_PATH=$HOME/.mujoco/mujoco200/bin:$LD_LIBRARY_PATH
export MUJOCO_PY_MJPRO_PATH=$HOME/.mujoco/mujoco200/
export MUJOCO_PY_MJKEY_PATH=$HOME/.mujoco/mujoco200/mjkey.txt

then I clone the dm_control suite from the Site. then I move to folder and then i install the package with command with pip3 install . to check whether it works for me or not
After installing with pip3 install .  i check the all available package in terminals 
Now When I try to run the code examples given with control suites then following error happen which is also title of ISSUE

alimuldal, 2020-05-27:
There are a few possible reasons why registration might fail:

* The path might be incorrect
* The contents of `mjkey.txt` might be invalid
* The license might be expired or otherwise invalid

You should see an error message logged by MuJoCo that should give more information about why the registration failed, e.g.
```
~$ MJKEY_PATH=/some/nonexistent/path/mjkey.txt \
  python -c "from dm_control import suite; suite.load('cartpole', 'swingup')"
CRITICAL:absl:Could not open activation key file /some/nonexistent/path/mjkey.txt
Traceback (most recent call last):
  ...

~$ MJKEY_PATH=~/.bashrc \
  python -c "from dm_control import suite; suite.load('cartpole', 'swingup')"
CRITICAL:absl:Invalid text in activation key
Traceback (most recent call last):
  ...
```

kashifzr, 2020-06-10:
alimuldal thanks. 
After finding the missing component like mujoco python file which is missing in system, it now install and works.

closed 2020-06-16 by alimuldal

FoRg1veness, 2020-06-26:
> alimuldal thanks.
> After finding the missing component like mujoco python file which is missing in system, it now install and works.

Hey how did you fixed it exactly? I am trying to find a solution but it seems difficult to find

alimuldal, 2020-06-26:
@FoRg1veness What do your logs show? In particular the `CRITICAL:absl:` part I mentioned above should tell you the reason why MuJoCo could not register the license.

jkren6, 2020-09-30:
> @FoRg1veness What do your logs show? In particular the `CRITICAL:absl:` part I mentioned above should tell you the reason why MuJoCo could not register the license.

Hi Alimuldal, 

Could u please have a look at my problem #158,  it shows "_CRITICAL:absl:Invalid activation key_"

I follow your suggestions in this issue, but I still didn't find the correct solution. I was obsessed with this problem for a few days ~~

Thanks very much!