google-deepmind/dm_control #152: How can I disable render in docker

nlnjnj, 2020-08-19:
As the title mentioned, I require running the env in a docker container, so I hope to disable render function.

alimuldal, 2020-08-24:
Have you tried using [software rendering](https://github.com/deepmind/dm_control#rendering)? OSMesa doesn't require a GPU or an X server, so it should work fine on a headless docker container. `dm_control` will automatically fall back on using OSMesa if it can't use either of the GPU backends, although you can also force it by setting the environment variable `MUJOCO_GL=osmesa`.

alimuldal, 2020-09-01:
Rendering contexts are created lazily when the `physics.contexts` property is accessed, so provided that you're not actually creating `Camera` instances or calling `physics.render()` yourself then in most cases there will be essentially zero cost associated with software rendering support. In any case, creating an OSMesa context is quite cheap in comparison to the cost of compiling the model.

alimuldal, 2020-09-25:
Closing this because I don't think there's a strong enough case for disabling rendering support entirely. Using the OSMesa backend and not calling `physics.render` should deliver almost the same performance since OSMesa context creation is fairly cheap, and only happens once per physics instance.

closed 2020-09-25 by alimuldal