google-deepmind/dm_control #158: dm_control.mujoco.wrapper.core.Error: Could not register license; CRITICAL:absl:Invalid activation key

jkren6, 2020-09-30:
Hi my friends,

dm_control suit is an awesome project which provides the most  useful and fruitful environments for people in this field.
  
I'm current training an Deep RL agent from pixel input. One month ago, I applied for a MuJoCo trial license, I install MuJoCo, mujoco_py and set it up with dm_control suit on my computer. It worked well on my computer which has a 1080ti GPU. 

Recently I applied for a MuJoCo student license through my university email, and replaced it with the previous trial license. In addition, I changed my 1080ti GPU to 2080ti GPU. After I did these, I ran the program, it says 

"_CRITICAL:absl:Invalid activation key

dm_control.mujoco.wrapper.core.Error: Could not register license._ "

The whole respond from the terminal is:
`(pytorch_1.6) jacky@asus:~/PycharmProjects/AE_A3C$ python3 train.py
[[V: not found
sh: 2: [
        : not found
[[V: not found
sh: 2: [
        : not found
[[V: not found
sh: 2: Syntax error: "(" unexpected
[[V: not found
sh: 2: Syntax error: "(" unexpected
CRITICAL:absl:Invalid activation key
Traceback (most recent call last):
  File "train.py", line 254, in <module>
    main()
  File "train.py", line 150, in main
    frame_skip=args.action_repeat
  File "/home/jacky/anaconda3/envs/pytorch_1.6/lib/python3.6/site-packages/dmc2gym-1.0.0-py3.6.egg/dmc2gym/__init__.py", line 52, in make
  File "/home/jacky/anaconda3/envs/pytorch_1.6/lib/python3.6/site-packages/gym/envs/registration.py", line 145, in make
    return registry.make(id, **kwargs)
  File "/home/jacky/anaconda3/envs/pytorch_1.6/lib/python3.6/site-packages/gym/envs/registration.py", line 90, in make
    env = spec.make(**kwargs)
  File "/home/jacky/anaconda3/envs/pytorch_1.6/lib/python3.6/site-packages/gym/envs/registration.py", line 60, in make
    env = cls(**_kwargs)
  File "/home/jacky/anaconda3/envs/pytorch_1.6/lib/python3.6/site-packages/dmc2gym-1.0.0-py3.6.egg/dmc2gym/wrappers.py", line 66, in __init__
  File "/home/jacky/anaconda3/envs/pytorch_1.6/lib/python3.6/site-packages/dm_control/suite/__init__.py", line 118, in load
    environment_kwargs, visualize_reward)
  File "/home/jacky/anaconda3/envs/pytorch_1.6/lib/python3.6/site-packages/dm_control/suite/__init__.py", line 152, in build_environment
    env = domain.SUITE[task_name](**task_kwargs)
  File "/home/jacky/anaconda3/envs/pytorch_1.6/lib/python3.6/site-packages/dm_control/suite/cheetah.py", line 49, in run
    physics = Physics.from_xml_string(*get_model_and_assets())
  File "/home/jacky/anaconda3/envs/pytorch_1.6/lib/python3.6/site-packages/dm_control/mujoco/engine.py", line 428, in from_xml_string
    model = wrapper.MjModel.from_xml_string(xml_string, assets=assets)
  File "/home/jacky/anaconda3/envs/pytorch_1.6/lib/python3.6/site-packages/dm_control/mujoco/wrapper/core.py", line 476, in from_xml_string
    model_ptr = _get_model_ptr_from_xml(xml_string=xml_string, assets=assets)
  File "/home/jacky/anaconda3/envs/pytorch_1.6/lib/python3.6/site-packages/dm_control/mujoco/wrapper/core.py", line 347, in _get_model_ptr_from_xml
    _maybe_register_license()
  File "/home/jacky/anaconda3/envs/pytorch_1.6/lib/python3.6/site-packages/dm_control/mujoco/wrapper/core.py", line 132, in _maybe_register_license
    raise Error("Could not register license.")
dm_control.mujoco.wrapper.core.Error: Could not register license.
`
Could u please give some help, any suggestion would be really appreciate :+1: 

Btw,  I'm pretty sure my license is valid because I checked the tutorial example from mujoco_py official website and it worked well.

jkren6, 2020-09-30:
For more clear, I attached the terminal window screen here:
[image]

jkren6, 2020-09-30:
I also referred to #145 , trying to find some solutions, but so sad, it didn't work.

saran-t, 2020-09-30:
MuJoCo itself is raising an "Invalid activation key" error which suggests that your key file itself is invalid. The only think I can suggest is to double check that the file is exactly as provided by Roboti, and if that continues not to work then to contact them directly for further help.

alimuldal, 2020-11-13:
Yeah it looks as though the keyfile is either corrupted or somehow invalid (is it possible that MuJoCo is trying to load the wrong path? did you override the `MJKEY_PATH` environment variable?). Also, not sure where those shell syntax errors coming from, i.e.
```
���[����[V: not found
sh: 2: [
: not found
```
These are nothing to do with our MuJoCo bindings, perhaps they hint at another problem?

hit618, 2020-11-24:
Hi @jkren6  , I am experiencing the same problem as you. My _mjkey.txt_ works fine on mujoco_py, but shows ‘_CRITICAL: absL :Invalid activation key_’ with dm_control. Have you found a solution? Thank you very much!

christsa, 2020-12-14:
Hi

I was experiencing the same problem and found the following fix:
In dm_control/mujoco/wrapper/core.py line 68, I set the following flag to True:
`_REGISTERED = True`

I believe it tries to register your MuJoCo license, which is already registered if you are using mujoco-py.

Hope this helps.

hit618, 2020-12-22:
@christsa Thank you, it works. The MuJoCo license is already registered.

alimuldal, 2020-12-23:
Thanks @christsa, that's very helpful!

I'd like to find a way to avoid this double-registration problem if possible, but I'm still finding it hard to reproduce the error you're seeing. It seems I can register the license repeatedly, both from `mujoco_py` and `dm_control`, without encountering any errors.

I am wondering whether this behaviour depends on the specific license type - are you using a "trial", "personal", or "institutional" MuJoCo license?

alimuldal, 2020-12-23:
Aha, I can reproduce the error by activating a trial license multiple times, but this doesn't seem to happen with institutional licenses (this seems to be a bug in MuJoCo itself).

There doesn't seem to be a way to query whether MuJoCo has already been activated, but I think we can work around the problem by unconditionally deactivating any active licenses before we call `mj_activate` (this doesn't seem to cause any errors).

alimuldal, 2021-01-05:
Fixed in d522fc327f9a6b280030ea2733c386a5d7f8fa1f

closed 2021-01-05 by alimuldal

SomeshDaga, 2021-05-22:
I still face this error when activating a license for the second time. It seems like calling `mjlib.mj_deactivate()` doesn't do anything for me. However, calling `mjlib.mj_activate()` a second time succeeds. I'm also on a MuJoCo trial license

famishedrover, 2023-07-12:
I know this issue is closed but I faced a similar issue. While registering dm_control looks for mjkey here : `result = mjlib.mj_activate(util.to_binary_string(path))`. In my case the mjkey.txt was within the `.mujoco/mujoco200_linux` directory, but dm_control was expecting it to be in `.mujoco/`. Maybe I missed installation instructions of mujoco200 but regardless when working with multiple mujoco versions I would assume keeping the key within that dir makes more sense.