google-deepmind/dm_control #159: manipulation vs suite and applying visual changes

melfm, 2020-10-06:
Is there a specific reason these two are separate entities for the time being? It seems like the way they are setup is different, and the way you create the environment etc.

Also, I found that applying visual changes (related to this [issue](https://github.com/deepmind/dm_control/issues/125)) is somewhat trickier inside the manipulation tasks, like the jaco_arm.xml
I assume there is a way to pass properties like texture (for the background colour etc) down to the engine, but at the moment, it doesnt seem like these properties are being passed down in these manipulation envs via the .xml file

Is there an easy/clean way to do this?

Thanks

alimuldal, 2020-10-26:
> I'm guessing this is a design choice to leave all these attributes outside of the xml and add them later in the code?

In the code snippet above you are still modifying the MJCF model before it has been compiled by MuJoCo. Here, `self.root_entity` is a `composer.Entity` object that represents the "arena" to which the arm (and sometimes other objects) are attached. `self.root_entity.mjcf_model` is an `mjcf.RootElement`, which is a Python representation of the MJCF model. As you've seen, you can programatically modify the model by setting attributes on this Python object. At the start of each episode, the `self.root_entity.mjcf_model` gets converted to XML and then compiled by MuJoCo in order to generate an `mjcf.Physics` instance, which provides the interface that allows us to step the simulation, render observations etc. The actual compilation happens [here](https://github.com/deepmind/dm_control/blob/master/dm_control/composer/environment.py#L236) inside `composer.Environment`, in case you're interested.

The main motivations for constructing the XML in this way are modularity and flexibility - you can quickly and easily generate new environment configurations by bolting together existing entities, and you can dynamically change the model in between episodes. I'd recommend taking a look at the MJCF and Composer sections of our [Colab tutorial](https://colab.sandbox.google.com/github/deepmind/dm_control/blob/master/tutorial.ipynb) if you haven't already - hopefully the examples should help to clarify.

> Is there a specific reason these two are separate entities for the time being? It seems like the way they are setup is different, and the way you create the environment etc.

The reasons are mainly historical - `dm_control.suite` predates both `dm_control.mjcf` and `dm_control.composer`, so for the most part they use static XML files rather than dynamically generating the XML in Python. If I was going to reimplement the control suite from scratch I would probably use `composer` and `mjcf`. However the current version using static XML files has served us quite well so far, and there's perhaps a bit less of a need for customisation for the control suite since it's intended to be a fixed set of benchmarks rather than a set of reusable components for building new tasks.

Does that answer your questions?

alimuldal, 2020-11-09:
Closing this for now, please comment if you have more questions

closed 2020-11-09 by alimuldal