google-deepmind/dm_control #160: Manipulator Env xpos and xmat are relative to a rotating reference frame

SomeshDaga, 2020-11-13:
It seems that "global" positions and orientations for the manipulator environment are reported with respect to the `arm_root` body which can rotate.

```python
from dm_control import suite

env = suite.load('manipulator', 'bring_ball')

# Set environment to arbitrary state
env.reset()

# Ensure states are up to date
env.physics.forward()

# Get 'global' position of the second joint
pos = env.physics.data.xpos[1]

print("Second Joint Pos: {}".format(pos))
```

The result is always `Second Joint Pos: [0.  0.  0.4]`, despite having different orientations for the first arm joint i.e. `arm_root`. This is a bit troublesome if we need to work with a fixed global coordinate system. I believe the `xpos` and `xmat` properties in Mujoco are meant to work with a fixed global coordinate system, leaving relative positions/orientations to the `qpos` property.

alimuldal, 2020-11-13:
The positions in `xpos` are in fact global - the reason that `env.physics.data.xpos[1]` is constant is because the `upper_arm` body is [fixed w.r.t. the world](https://github.com/deepmind/dm_control/blob/master/dm_control/suite/manipulator.xml#L69). Note that there's no `<joint/>` element between `<worldbody>` and `<body name="upper_arm" ...>`. `middle_arm` is the first moveable body, and you'll see that its position changes as the joint angles in `qpos` change.

alimuldal, 2020-11-13:
That is correct behaviour. The reason the global position of the `upper_arm`  body doesn't change is because it rotates about its center. If you look at `xmat[1]` you'll see that the global orientation of the body changes when `qpos[0]` changes.

closed 2020-11-13 by alimuldal

SomeshDaga, 2020-11-13:
@alimuldal Thanks for the quick reply. You are indeed correct, that the `upper_arm` is fixed w.r.t the world body. That was a bad example on my part. What I'm getting confused with is that the `xpos` value for the subsequent joints does not make sense to me. As an example, this is the configuration of my arm:

[image]

And here are the `xpos` values:

```
[[ 0.00000000e+00  0.00000000e+00  0.00000000e+00]
 [ 0.00000000e+00  0.00000000e+00  4.00000000e-01]
 [ 8.80425918e-03  0.00000000e+00  5.79784552e-01]
 [ 9.37152820e-02  0.00000000e+00  7.03437769e-01]
 [ 1.70247045e-01  0.00000000e+00  6.11010018e-01]
 [ 1.27753949e-01  0.00000000e+00  5.31673162e-01]
 [ 1.22554878e-01  0.00000000e+00  5.85505955e-01]
 [ 9.93270283e-02  0.00000000e+00  5.40113588e-01]
 [ 1.75446116e-01  0.00000000e+00  5.57177224e-01]
 [ 1.50533166e-01  6.24340122e-18  5.12687395e-01]
 [ 1.70241243e-01  0.00000000e+00  6.78188470e-01]
 [-1.16431474e-01  1.00000000e-03  2.15432715e-01]]
```

I'm not sure why the x-coordinate values are positive (w.r.t to what I believe is the global reference frame, shown in the image). Also, the z-coordinate in this example, becomes smaller after the 4th row, but the arm seems to be pointed up, with the z-coordinate rising between all subsequent joints.

SomeshDaga, 2020-11-13:
I've realized the folly of my ways. It seems like the values I read after a `env.reset()` and not maintained after calling `viewer.launch(env)`, resulting in the discrepancy. So, the values I read directly after the `reset()` call were disagreeing with the visuals generated by `viewer.launch`

alimuldal, 2020-11-16:
Yeah, I think `viewer.launch(env)` calls `env.reset()` internally, so you'll get a different random initialization