google-deepmind/dm_control #162: Colab error from this link An introductory tutorial for this package is available as a Colaboratory notebook:

BBCBasic, 2020-12-30:
There was an error loading this notebook. Ensure that the file is accessible and try again.

JSON.parse: unexpected character at line 40 column 7 of the JSON data
https://github.com/deepmind/dm_control/blob/master/tutorial.ipynb

JSON.parse: unexpected character at line 40 column 7 of the JSON data
c/<@https://colab.research.google.com/v2/external/external_polymer_binary.js?vrz=colab-20201222-085609-RC00_348636243:1315:452
Ea@https://colab.research.google.com/v2/external/external_polymer_binary.js?vrz=colab-20201222-085609-RC00_348636243:19:336
Ca.prototype.next_@https://colab.research.google.com/v2/external/external_polymer_binary.js?vrz=colab-20201222-085609-RC00_348636243:17:503
Ga/this.next@https://colab.research.google.com/v2/external/external_polymer_binary.js?vrz=colab-20201222-085609-RC00_348636243:20:206
f@https://colab.research.google.com/v2/external/external_polymer_binary.js?vrz=colab-20201222-085609-RC00_348636243:70:101

alimuldal, 2020-12-30:
Thanks for reporting, should be an easy fix. I could have sworn that Colab's JSON parser used to allow "#"s...

closed 2020-12-31 by alimuldal

BBCBasic, 2020-12-31:
Thanks, working now. Cheers Rob.

Closed #162<https://github.com/deepmind/dm_control/issues/162> via d522fc3<https://github.com/deepmind/dm_control/commit/d522fc327f9a6b280030ea2733c386a5d7f8fa1f>.

commit d522fc32 (2020-12-31): Replace bare `#`  characters in `tutorial.ipynb` (fixes #162)

PiperOrigin-RevId: 349668059
Change-Id: Id5ad8af6d010f0534976ecc7ca951a8500ed52ce