google-deepmind/dm_control #166: missed files

yueyue-yue, 2021-01-26:
from dm_control.mujoco.wrapper.mjbindings import constants
from dm_control.mujoco.wrapper.mjbindings import enums
from dm_control.mujoco.wrapper.mjbindings import sizes
from dm_control.mujoco.wrapper.mjbindings import types
from dm_control.mujoco.wrapper.mjbindings import wrappers

missed files 'constants', 'enums', 'sizes', 'types', 'wrappers'

yueyue-yue, 2021-01-26:
location: dm_control/mujoco/wrapper/mjbindings/__init__.py

yueyue-yue, 2021-01-26:
ImportError: 
cannot import name 'constants' from partially initialized module 'dm_control.mujoco.wrapper.mjbindings' (most likely due to a circular import) 
(/home/xxx/dm_control/mujoco/wrapper/mjbindings/__init__.py)

yueyue-yue, 2021-01-26:
When running files in deep mind research/catch-carry, this mistake always occur.

alimuldal, 2021-01-26:
Possible duplicate of  #6. Those source files are normally generated automatically when you `pip install <path to dm_control>`.

yueyue-yue, 2021-01-27:
Thank you and I have fixed this successfully. But sorry to brother again, a new problem emerged:
'''
WARNING:absl:mjbindings failed to import mjlib and other functions. libmujoco.so may not be accessible.
...
...
OSError: /home/user/.mujoco/mujoco200_linux/bin/libmujoco200.so: undefined symbol: __glewBindBuffer
'''
My mujoco engine can work well seperately and the location of the mujoco files are located at default paths.

alimuldal, 2021-01-27:
That error means that libglew is not being found - perhaps you need to install it? See the instructions here: https://github.com/deepmind/dm_control#rendering

closed 2021-01-27 by alimuldal

mingyucai, 2021-05-06:
when I do "from dm_control import suite" I got the error: 
'''
cannot import name 'constants' from partially initialized module 'dm_control.mujoco.wrapper.mjbindings' (most likely due to a circular import) (/home/mingyu/dm_control/dm_control/mujoco/wrapper/mjbindings/__init__.py)
'''
My system is Ubuntu 20 and I have installed libglew2.1, which is not libglew2.0. I don't know the issues.

mingyucai, 2021-05-06:
I also strictly follow the instruction 'pip3 install . --user'. But still can got the error
'''
ImportError:
cannot import name 'constants' from partially initialized module 'dm_control.mujoco.wrapper.mjbindings' (most likely due to a circular import)
(/home/xxx/dm_control/mujoco/wrapper/mjbindings/init.py)
""

mingyucai, 2021-05-06:
I solved the issue. I found after pip install, the python locate the dm_control module in the download folder, not the python package folder.

mingyucai, 2021-05-06:
Thanks for the great source.

yueyue-yue, 2021-05-06:
Thanks a lot!!!
Best Wishes!!

        - External Email -