google-deepmind/dm_control #167: difference between "CheetahRun-v0" on DM vs "half-cheetah-v2" on Mujuco

randyhouse, 2021-02-09:
What is the difference between "CheetahRun-v0" on DM vs "half-cheetah-v2" on Mujuco?

From what I can tell, do they only differ by the reward functions?

alimuldal, 2021-02-09:
I recognise "HalfCheetah-v2" as an OpenAI Gym level, but I'm not quite sure what "CheetahRun-v0" refers to (that's not a naming convention we use within `dm_control`). Are you asking about `dm_control.suite.load(domain_name='cheetah', task_name='run')`?

randyhouse, 2021-02-09:
> I recognise "HalfCheetah-v2" as an OpenAI Gym level, but I'm not quite sure what "CheetahRun-v0" refers to (that's not a naming convention we use within `dm_control`). Are you asking about `dm_control.suite.load(domain_name='cheetah', task_name='run')`?

Yes. I am talking about DM-control suite. Do you know the difference between the two?

alimuldal, 2021-02-15:
The MJCF models are extremely similar - I believe they're both based on [this paper](https://doi.org/10.1016/j.neunet.2009.05.011). Aside from the aesthetics, the only notable difference I spotted was that the gym version uses non-default values for `solimp` and `solimplimit`. These parameters influence the constraint forces for contacts and joint limits respectively [(see here)](http://mujoco.org/book/modeling.html#CSolver). Contacts and joint constraints would be slightly "softer" in the gym version than in ours.

Most of the differences are on the Python side. As you already mentioned, the two environments differ in terms of their reward functions - ours is a linear function of speed and is clipped between 0 and 1, whereas theirs is unbounded and also incorporates a control cost. There are also some differences in terms of randomisation - we sample our initial joint angles uniformly within each joint's range and then step the simulation for a few steps to allow things to settle, whereas the gym version adds a small amount of uniform noise to the initial pose and a small velocity perturbation. Our episodes also have a fixed duration of 1000 steps.

closed 2021-02-19 by alimuldal