google-deepmind/dm_control #171: Failed Installing dm_control with Bazel loaded

DarrenZhang01, 2021-02-26:
As titled, I have already loaded Bazel.

The error is as follows. Any insights would be appreciated.

```
Building wheels for collected packages: labmaze
  Building wheel for labmaze (setup.py) ... error
  ERROR: Command errored out with exit status 1:
   command: /home/zhibo/mujoco_env/bin/python -u -c 'import sys, setuptools, tokenize; sys.argv[0] = '"'"'/tmp/pip-install-8952tcao/labmaze_a6a968ff04254ef58eca0e34b0f3433e/setup.py'"'"'; __file__='"'"'/tmp/pip-install-8952tcao/labmaze_a6a968ff04254ef58eca0e34b0f3433e/setup.py'"'"';f=getattr(tokenize, '"'"'open'"'"', open)(__file__);code=f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();exec(compile(code, __file__, '"'"'exec'"'"'))' bdist_wheel -d /tmp/pip-wheel-rgis3lee
       cwd: /tmp/pip-install-8952tcao/labmaze_a6a968ff04254ef58eca0e34b0f3433e/
  Complete output (126 lines):
  running bdist_wheel
  running build
  running build_py
  creating build
  creating build/lib.linux-x86_64-3.6
  creating build/lib.linux-x86_64-3.6/labmaze
  copying labmaze/__init__.py -> build/lib.linux-x86_64-3.6/labmaze
  copying labmaze/base.py -> build/lib.linux-x86_64-3.6/labmaze
  copying labmaze/defaults.py -> build/lib.linux-x86_64-3.6/labmaze
  copying labmaze/fixed_maze.py -> build/lib.linux-x86_64-3.6/labmaze
  copying labmaze/fixed_maze_test.py -> build/lib.linux-x86_64-3.6/labmaze
  copying labmaze/random_maze.py -> build/lib.linux-x86_64-3.6/labmaze
  copying labmaze/random_maze_test.py -> build/lib.linux-x86_64-3.6/labmaze
  copying labmaze/text_grid.py -> build/lib.linux-x86_64-3.6/labmaze
  copying labmaze/text_grid_test.py -> build/lib.linux-x86_64-3.6/labmaze
  creating build/lib.linux-x86_64-3.6/labmaze/assets
  copying labmaze/assets/__init__.py -> build/lib.linux-x86_64-3.6/labmaze/assets
  copying labmaze/assets/assets_test.py -> build/lib.linux-x86_64-3.6/labmaze/assets
  creating build/lib.linux-x86_64-3.6/labmaze/cc
  copying labmaze/cc/__init__.py -> build/lib.linux-x86_64-3.6/labmaze/cc
  creating build/lib.linux-x86_64-3.6/labmaze/cc/python
  copying labmaze/cc/python/__init__.py -> build/lib.linux-x86_64-3.6/labmaze/cc/python
  creating build/lib.linux-x86_64-3.6/labmaze/assets/style_03
  copying labmaze/assets/style_03/floor_green_d.png -> build/lib.linux-x86_64-3.6/labmaze/assets/style_03
  creating build/lib.linux-x86_64-3.6/labmaze/assets/style_01
  copying labmaze/assets/style_01/wall_cerise_d.png -> build/lib.linux-x86_64-3.6/labmaze/assets/style_01
  copying labmaze/assets/style_01/wall_red_bright_d.png -> build/lib.linux-x86_64-3.6/labmaze/assets/style_01
  creating build/lib.linux-x86_64-3.6/labmaze/assets/style_04
  copying labmaze/assets/style_04/wall_cerise_d.png -> build/lib.linux-x86_64-3.6/labmaze/assets/style_04
  creating build/lib.linux-x86_64-3.6/labmaze/assets/style_05
  copying labmaze/assets/style_05/wall_yellow_d.png -> build/lib.linux-x86_64-3.6/labmaze/assets/style_05
  copying labmaze/assets/style_05/floor_light1_m.png -> build/lib.linux-x86_64-3.6/labmaze/assets/style_05
  copying labmaze/assets/style_01/wall_green_bright_d.png -> build/lib.linux-x86_64-3.6/labmaze/assets/style_01
  copying labmaze/assets/style_04/wall_green_d.png -> build/lib.linux-x86_64-3.6/labmaze/assets/style_04
  copying labmaze/assets/style_01/wall_light_m.png -> build/lib.linux-x86_64-3.6/labmaze/assets/style_01
  copying labmaze/assets/style_04/wall_red_d.png -> build/lib.linux-x86_64-3.6/labmaze/assets/style_04
  creating build/lib.linux-x86_64-3.6/labmaze/assets/sky_01
  copying labmaze/assets/sky_01/ft.png -> build/lib.linux-x86_64-3.6/labmaze/assets/sky_01
  creating build/lib.linux-x86_64-3.6/labmaze/assets/style_02
  copying labmaze/assets/style_02/wall_dblue_d.png -> build/lib.linux-x86_64-3.6/labmaze/assets/style_02
  copying labmaze/assets/style_03/floor_blue_bright_d.png -> build/lib.linux-x86_64-3.6/labmaze/assets/style_03
  copying labmaze/assets/style_01/floor_orange_d.png -> build/lib.linux-x86_64-3.6/labmaze/assets/style_01
  creating build/lib.linux-x86_64-3.6/labmaze/assets/sky_02
  copying labmaze/assets/sky_02/rt.png -> build/lib.linux-x86_64-3.6/labmaze/assets/sky_02
  copying labmaze/assets/style_03/wall_gray_d.png -> build/lib.linux-x86_64-3.6/labmaze/assets/style_03
  copying labmaze/assets/style_03/wall_cyan_d.png -> build/lib.linux-x86_64-3.6/labmaze/assets/style_03
  copying labmaze/assets/style_04/floor_orange_d.png -> build/lib.linux-x86_64-3.6/labmaze/assets/style_04
  copying labmaze/assets/style_04/wall_red_bright_d.png -> build/lib.linux-x86_64-3.6/labmaze/assets/style_04
  copying labmaze/assets/style_05/wall_red_d.png -> build/lib.linux-x86_64-3.6/labmaze/assets/style_05
  copying labmaze/assets/style_03/floor_blue_d.png -> build/lib.linux-x86_64-3.6/labmaze/assets/style_03
  copying labmaze/assets/style_01/wall_purple_d.png -> build/lib.linux-x86_64-3.6/labmaze/assets/style_01
  copying labmaze/assets/style_01/wall_green_d.png -> build/lib.linux-x86_64-3.6/labmaze/assets/style_01
  copying labmaze/assets/sky_02/lf.png -> build/lib.linux-x86_64-3.6/labmaze/assets/sky_02
  creating build/lib.linux-x86_64-3.6/labmaze/assets/sky_03
  copying labmaze/assets/sky_03/rt.png -> build/lib.linux-x86_64-3.6/labmaze/assets/sky_03
  copying labmaze/assets/sky_02/up.png -> build/lib.linux-x86_64-3.6/labmaze/assets/sky_02
  copying labmaze/assets/style_01/floor_blue_bright_d.png -> build/lib.linux-x86_64-3.6/labmaze/assets/style_01
  copying labmaze/assets/style_04/floor_blue_bright_d.png -> build/lib.linux-x86_64-3.6/labmaze/assets/style_04
  copying labmaze/assets/style_05/floor_light2_m.png -> build/lib.linux-x86_64-3.6/labmaze/assets/style_05
[... 181 lines cut ...]
    copying labmaze/assets/style_02/wall_yellow_d.png -> build/lib.linux-x86_64-3.6/labmaze/assets/style_02
    copying labmaze/assets/style_02/wall_blue_d.png -> build/lib.linux-x86_64-3.6/labmaze/assets/style_02
    copying labmaze/assets/style_04/floor_cyan_d.png -> build/lib.linux-x86_64-3.6/labmaze/assets/style_04
    copying labmaze/assets/style_05/floor_orange_d.png -> build/lib.linux-x86_64-3.6/labmaze/assets/style_05
    copying labmaze/assets/sky_03/up.png -> build/lib.linux-x86_64-3.6/labmaze/assets/sky_03
    copying labmaze/assets/style_03/floor_light1_m.png -> build/lib.linux-x86_64-3.6/labmaze/assets/style_03
    copying labmaze/assets/sky_02/rt.png -> build/lib.linux-x86_64-3.6/labmaze/assets/sky_02
    copying labmaze/assets/style_03/floor_green_bright_d.png -> build/lib.linux-x86_64-3.6/labmaze/assets/style_03
    copying labmaze/assets/style_04/floor_blue_d.png -> build/lib.linux-x86_64-3.6/labmaze/assets/style_04
    copying labmaze/assets/sky_03/dn.png -> build/lib.linux-x86_64-3.6/labmaze/assets/sky_03
    copying labmaze/assets/style_02/wall_blue_bright_d.png -> build/lib.linux-x86_64-3.6/labmaze/assets/style_02
    copying labmaze/assets/style_02/wall_yellow_bright_d.png -> build/lib.linux-x86_64-3.6/labmaze/assets/style_02
    copying labmaze/assets/style_02/wall_dblue_d.png -> build/lib.linux-x86_64-3.6/labmaze/assets/style_02
    copying labmaze/assets/style_03/wall_cyan_d.png -> build/lib.linux-x86_64-3.6/labmaze/assets/style_03
    copying labmaze/assets/style_01/wall_green_d.png -> build/lib.linux-x86_64-3.6/labmaze/assets/style_01
    running build_ext
    bazel build //labmaze/cc/python:_defaults --symlink_prefix=build/temp.linux-x86_64-3.6/bazel- --compilation_mode=opt
    Starting local Bazel server and connecting to it...
    Server crashed during startup. Now printing /home/zhibo/.cache/bazel/_bazel_zhibo/76b1e240ad9500159819ba8ac7d2079b/server/jvm.out
    Error occurred during initialization of VM
    Could not allocate metaspace: 1073741824 bytes
    error: command 'bazel' failed with exit status 37
    ----------------------------------------
ERROR: Command errored out with exit status 1: /home/zhibo/mujoco_env/bin/python -u -c 'import sys, setuptools, tokenize; sys.argv[0] = '"'"'/tmp/pip-install-8952tcao/labmaze_a6a968ff04254ef58eca0e34b0f3433e/setup.py'"'"'; __file__='"'"'/tmp/pip-install-8952tcao/labmaze_a6a968ff04254ef58eca0e34b0f3433e/setup.py'"'"';f=getattr(tokenize, '"'"'open'"'"', open)(__file__);code=f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();exec(compile(code, __file__, '"'"'exec'"'"'))' install --record /tmp/pip-record-smfbrl2n/install-record.txt --single-version-externally-managed --compile --install-headers /home/zhibo/mujoco_env/include/site/python3.6/labmaze Check the logs for full command output.
```

alimuldal, 2021-02-27:
This looks more like a `bazel` issue (or really a Java issue) rather than being specific to `dm_control`. I believe this error:
```
Could not allocate metaspace: 1073741824 bytes
```
means that the JVM is trying and failing to grab a chunk of system memory on startup (1073741824 bytes = 1 GiB, which I think is the default amount of memory it requests for storing object metadata). My guess is that you probably just don't have enough free memory available.

You can either try to free up some system memory, or you could experiment by setting environment variables to control how much memory JVM requests, e.g.
```shell
export BAZEL_JAVAC_OPTS="-XX:CompressedClassSpaceSize=<size>m"
```
where `size` is in megabytes. See [here](https://docs.oracle.com/javase/10/gctuning/other-considerations.htm#JSGCT-GUID-B29C9153-3530-4C15-9154-E74F44E3DAD9) for more info.

DarrenZhang01, 2021-02-27:
> This looks more like a `bazel` issue (or really a Java issue) rather than being specific to `dm_control`. I believe this error:
>
> ```
> Could not allocate metaspace: 1073741824 bytes
> ```
>
> means that [...]

Thanks very much for your reply and pointer, Alistair! 

I came across this issue on Compute Canada. I followed the command you provided and the instructions under https://docs.computecanada.ca/wiki/Java. I ran `java` or `javac` command and they worked totally fine. But when I tried to install dm_control, the Bazel started to complain ```Unable to setup metadata virtual space list.```, as follows.

```
Building wheels for collected packages: labmaze
  Building wheel for labmaze (setup.py) ... error
  ERROR: Command errored out with exit status 1:
   command: /home/zhibo/mujoco_env/bin/python -u -c 'import sys, setuptools, tokenize; sys.argv[0] = '"'"'/tmp/pip-install-pv31h8vj/labmaze_73ddfbc9e17449eb8524536b107dc52e/setup.py'"'"'; __file__='"'"'/tmp/pip-install-pv31h8vj/labmaze_73ddfbc9e17449eb8524536b107dc52e/setup.py'"'"';f=getattr(tokenize, '"'"'open'"'"', open)(__file__);code=f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();exec(compile(code, __file__, '"'"'exec'"'"'))' bdist_wheel -d /tmp/pip-wheel-rpe_stse
       cwd: /tmp/pip-install-pv31h8vj/labmaze_73ddfbc9e17449eb8524536b107dc52e/
  Complete output (128 lines):
  running bdist_wheel
  running build
  running build_py
  creating build
  creating build/lib.linux-x86_64-3.6
  creating build/lib.linux-x86_64-3.6/labmaze
  copying labmaze/__init__.py -> build/lib.linux-x86_64-3.6/labmaze
  copying labmaze/base.py -> build/lib.linux-x86_64-3.6/labmaze
  copying labmaze/defaults.py -> build/lib.linux-x86_64-3.6/labmaze
  copying labmaze/fixed_maze.py -> build/lib.linux-x86_64-3.6/labmaze
  copying labmaze/fixed_maze_test.py -> build/lib.linux-x86_64-3.6/labmaze
  copying labmaze/random_maze.py -> build/lib.linux-x86_64-3.6/labmaze
  copying labmaze/random_maze_test.py -> build/lib.linux-x86_64-3.6/labmaze
  copying labmaze/text_grid.py -> build/lib.linux-x86_64-3.6/labmaze
  copying labmaze/text_grid_test.py -> build/lib.linux-x86_64-3.6/labmaze
  creating build/lib.linux-x86_64-3.6/labmaze/assets
  copying labmaze/assets/__init__.py -> build/lib.linux-x86_64-3.6/labmaze/assets
  copying labmaze/assets/assets_test.py -> build/lib.linux-x86_64-3.6/labmaze/assets
  creating build/lib.linux-x86_64-3.6/labmaze/cc
  copying labmaze/cc/__init__.py -> build/lib.linux-x86_64-3.6/labmaze/cc
  creating build/lib.linux-x86_64-3.6/labmaze/cc/python
  copying labmaze/cc/python/__init__.py -> build/lib.linux-x86_64-3.6/labmaze/cc/python
  creating build/lib.linux-x86_64-3.6/labmaze/assets/sky_02
  copying labmaze/assets/sky_02/up.png -> build/lib.linux-x86_64-3.6/labmaze/assets/sky_02
  creating build/lib.linux-x86_64-3.6/labmaze/assets/style_01
  copying labmaze/assets/style_01/wall_blue_d.png -> build/lib.linux-x86_64-3.6/labmaze/assets/style_01
  copying labmaze/assets/style_01/floor_light_m.png -> build/lib.linux-x86_64-3.6/labmaze/assets/style_01
  creating build/lib.linux-x86_64-3.6/labmaze/assets/style_05
  copying labmaze/assets/style_05/floor_blue_d.png -> build/lib.linux-x86_64-3.6/labmaze/assets/style_05
  copying labmaze/assets/style_05/floor_orange_bright_d.png -> build/lib.linux-x86_64-3.6/labmaze/assets/style_05
  creating build/lib.linux-x86_64-3.6/labmaze/assets/sky_01
  copying labmaze/assets/sky_01/rt.png -> build/lib.linux-x86_64-3.6/labmaze/assets/sky_01
  creating build/lib.linux-x86_64-3.6/labmaze/assets/style_02
  copying labmaze/assets/style_02/wall_lgreen_d.png -> build/lib.linux-x86_64-3.6/labmaze/assets/style_02
  copying labmaze/assets/style_01/wall_light_m.png -> build/lib.linux-x86_64-3.6/labmaze/assets/style_01
  creating build/lib.linux-x86_64-3.6/labmaze/assets/style_04
  copying labmaze/assets/style_04/floor_cyan_d.png -> build/lib.linux-x86_64-3.6/labmaze/assets/style_04
  copying labmaze/assets/style_01/floor_orange_bright_d.png -> build/lib.linux-x86_64-3.6/labmaze/assets/style_01
  creating build/lib.linux-x86_64-3.6/labmaze/assets/sky_03
  copying labmaze/assets/sky_03/ft.png -> build/lib.linux-x86_64-3.6/labmaze/assets/sky_03
  creating build/lib.linux-x86_64-3.6/labmaze/assets/style_03
  copying labmaze/assets/style_03/wall_cyan_d.png -> build/lib.linux-x86_64-3.6/labmaze/assets/style_03
  copying labmaze/assets/style_01/floor_blue_team_d.png -> build/lib.linux-x86_64-3.6/labmaze/assets/style_01
  copying labmaze/assets/style_04/floor_blue_d.png -> build/lib.linux-x86_64-3.6/labmaze/assets/style_04
  copying labmaze/assets/style_01/wall_red_d.png -> build/lib.linux-x86_64-3.6/labmaze/assets/style_01
  copying labmaze/assets/style_02/floor_green_d.png -> build/lib.linux-x86_64-3.6/labmaze/assets/style_02
  copying labmaze/assets/style_03/wall_gray_bright_d.png -> build/lib.linux-x86_64-3.6/labmaze/assets/style_03
  copying labmaze/assets/style_05/floor_light1_m.png -> build/lib.linux-x86_64-3.6/labmaze/assets/style_05
  copying labmaze/assets/style_05/floor_blue_bright_d.png -> build/lib.linux-x86_64-3.6/labmaze/assets/style_05
  copying labmaze/assets/sky_02/lf.png -> build/lib.linux-x86_64-3.6/labmaze/assets/sky_02
  copying labmaze/assets/style_05/wall_yellow_d.png -> build/lib.linux-x86_64-3.6/labmaze/assets/style_05
[... 193 lines cut ...]
    copying labmaze/assets/style_03/floor_green_bright_d.png -> build/lib.linux-x86_64-3.6/labmaze/assets/style_03
[... cut ...] 7dc52e/setup.py'"'"'; __file__='"'"'/tmp/pip-install-pv31h8vj/labmaze_73ddfbc9e17449eb8524536b107dc52e/setup.py'"'"';f=getattr(tokenize, '"'"'open'"'"', open)(__file__);code=f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();exec(compile(code, __file__, '"'"'exec'"'"'))' install --record /tmp/pip-record-8d6mqala/install-record.txt --single-version-externally-managed --compile --install-headers /home/zhibo/mujoco_env/include/site/python3.6/labmaze Check the logs for full command output.
```

alimuldal, 2021-02-27:
That probably means you're setting the limit too low. You may need to experiment a bit in order to find a value that's large enough for the build to succeed, but small enough that it will fit in memory. It's possible that you just don't have enough memory to complete the build. Is there a particular reason why you're building `labmaze` from source? There are pre-built wheels available, so usually this isn't necessary.

In any case I'm going to close this issue since it's definitely not a problem with `dm_control` (I see that you've already opened a [similar issue](https://github.com/deepmind/labmaze/issues/1) in the `labmaze` repo, although the problem is really with your build environment rather than with `labmaze` itself).

closed 2021-02-27 by alimuldal

liovn, 2023-03-03:
May I ask if you have solved this problem? I have encountered the same mistake as you. If you can give me some advice, I would be very grateful

saran-t, 2023-03-03:
We don't ship prebuilt labmaze binaries for Pythons 3.6 or older, since these versions are beyond their end-of-life.