google-deepmind/dm_control #172: Tutorial Colab raises GLEW initalization error when rendering

LennartAig, 2021-02-27:
The tutorial **colab** notebook crashes as soon as rendering is called:

`from dm_control import suite
  env = suite.load('cartpole', 'swingup')
  pixels = env.physics.render()`

The following error rises: 
`CRITICAL:absl:GLEW initalization error: Missing GL version `

I tried different instances with and without GPU.
After some research through the issues here and https://github.com/openai/mujoco-py/issues/268, I still wasn't able to solve it.

Thanks for your help.

alimuldal, 2021-02-27:
Are you using a local Colab kernel, or one that's running on Google Cloud?

If it's a local kernel, are you able to load and render an environment outside of a Colab, e.g. like in [this example](https://github.com/deepmind/dm_control/blob/master/dm_control/mujoco/README.md)?

Have you read the instructions [here](https://github.com/deepmind/dm_control#rendering) about installing rendering dependencies? Are you overriding the `MUJOCO_GL=` environment variable?

LennartAig, 2021-02-27:
Thanks for the quick reply.

I am using a cloud kernel (to use the provided GPU) executing the [notebook](https://colab.research.google.com/github/deepmind/dm_control/blob/master/tutorial.ipynb) provided in the Readme. 
In there, by default, the rendering dependencies are installed on the current instance and the environment variable is overridden:
`%env MUJOCO_GL=osmesa`. 

Is the provided notebook not suited for google colab, or do I need some specific instance?

alimuldal, 2021-02-27:
Thanks, that is useful information - I am able to reproduce the problem. I am not yet sure what has changed, as this certainly used to work on cloud kernels. Can you run a local kernel as a temporary workaround?

LennartAig, 2021-02-27:
On my local setup, everything works as intended, but not a lot of compute unfortunately. 

Thanks a lot for looking in to this, at least I know now, that it is not possible to use in Colab at the moment. Feel free to close the issue. It would be nice if this is resolved at some point, especially for students. But I know that this is not really only a dm_control related issue.

alimuldal, 2021-02-27:
Hmm, debugging within a Colab VM is a bit painful. What I'm trying at the moment is logging symbol binding events:
```python
!pip install dm_control

%env MUJOCO_GL=osmesa
%env LD_DEBUG=bindings

!python -c "from dm_control import suite; suite.load('cartpole', 'swingup').physics.render(); print('success')"
```
The output is very long, but at the end we see some symbol lookup errors relating to `libOSMesa` which seem potentially relevant:
```
     14655:	calling init: /usr/lib/python3.7/lib-dynload/_queue.cpython-37m-x86_64-linux-gnu.so
     14655:	
     14655:	binding file /usr/lib/python3.7/lib-dynload/_queue.cpython-37m-x86_64-linux-gnu.so [0] to /usr/lib/python3.7/lib-dynload/_queue.cpython-37m-x86_64-linux-gnu.so [0]: normal symbol `PyInit__queue'
     14655:	binding file /lib/x86_64-linux-gnu/libpthread.so.0 [0] to /usr/lib/x86_64-linux-gnu/libtcmalloc.so.4 [0]: normal symbol `free' [GLIBC_2.2.5]
     14655:	binding file /lib/x86_64-linux-gnu/libpthread.so.0 [0] to /lib64/ld-linux-x86-64.so.2 [0]: normal symbol `_dl_allocate_tls_init' [GLIBC_PRIVATE]
     14655:	/usr/lib/x86_64-linux-gnu/libOSMesa.so.8: error: symbol lookup error: undefined symbol: FunctionType (fatal)
     14655:	binding file /usr/lib/x86_64-linux-gnu/libOSMesa.so.8 [0] to /usr/lib/x86_64-linux-gnu/libOSMesa.so.8 [0]: normal symbol `OSMesaCreateContextExt'
     14655:	binding file /lib/x86_64-linux-gnu/libpthread.so.0 [0] to /lib/x86_64-linux-gnu/libc.so.6 [0]: normal symbol `strlen' [GLIBC_2.2.5]
     14655:	binding file /lib/x86_64-linux-gnu/libpthread.so.0 [0] to /lib/x86_64-linux-gnu/libc.so.6 [0]: normal symbol `prctl' [GLIBC_2.2.5]
     14655:	/usr/lib/x86_64-linux-gnu/libOSMesa.so.8: error: symbol lookup error: undefined symbol: FunctionType (fatal)
     14655:	binding file /usr/lib/x86_64-linux-gnu/libOSMesa.so.8 [0] to /usr/lib/x86_64-linux-gnu/libOSMesa.so.8 [0]: normal symbol `OSMesaMakeCurrent'
```

LennartAig, 2021-02-27:
Setting the mujoco gl environment variable to "Headless" hardware rendering solved the issue for me using a colab cloud GPU instance. I see no potential downside compared to osmesa. 

`%env MUJOCO_GL=egl`

alimuldal, 2021-02-27:
I'm glad that works for you. It does mean that you'd be sharing the GPU if you're using it for training/inference as well as rendering, which may or may not be a bad thing. I will leave this issue open anyway, since we should ideally support non-hardware rendering as well.