google-deepmind/dm_control #183: Do you plan to add support for open simulator ?

jrabary, 2021-05-04:
dm_control is becoming a reference as far as evaluating RL benchmark is concerned. Unfortunately it is based on a Mujoco  which is a closed and not easily accessible for every one. Furthermore, the development of MuJoCo seems to be dead, last release 3 years ago, no support for such an expensive product.
With the research reproducibility in mind, why did Deepmind not choose an open simulator as a backend for dm_control ? Or add possibilities to use several backend ?

alimuldal, 2021-05-06:
No, this is not something we plan on doing in the foreseeable future. For better and worse, the core of `dm_control` is very closely modeled around MuJoCo's APIs and XML schema. Adding support for alternative physics engines would require a substantial rewrite (I suspect we'd probably need to start again more or less from scratch), and that's not something we have capacity for. MuJoCo is also a good fit for our current needs, and it's a tool that we're developed a lot of familiarity and expertise with over the years. Hopefully you can understand why we might want to continue to build on what we have now, rather than investing a large amount of time and effort implementing support for new backends.

closed 2021-05-06 by alimuldal

jrabary, 2021-05-09:
Thank you for answering my question. I totally understand your concern but the problem is dm_control becoming a de facto tool for benchmarking RL algorithms. Most of the major publications related to RL and robotics use dm_control. And it is difficult for us to reproduce those papers and to compare our methods with the state of the art without requiring dm_control and MuJoCo. Furthermore, Deepmind as a major actor in this field did a lot of advertising for dm_control. So why do you advertise a tool that your are the only one that is able to use it ?
The main issue for us is that it is difficult to buy a license for MuJoCo. It is not even the price. It is because nobody answers at Roboti, no feedback and no support for people that are willing to buy  several thousand dollars of license.
I know you are not roboti but as a big actor in this field that advocates reproductible research it can be a good this to avoid proprietary tools in the future. Things that Google and Deepmind do very well elsewhere. So why did you choose MuJoCo instead of pybullet for instance ?