google-deepmind/dm_control #184: Difference in environment rendering between dm_control and gym

ArgyChris, 2021-05-14:
Hello!
I am trying to compare the reacher in dm_control against the default reacher in gym (Reacher-v2) 
When I render the observations to an image, their colors are different, lighter for dm_control (top image), and darker for the default gym (bottom image). 
I used gym's reacher.xml to configure dm_control's environment to be the same and have a common appearance,
 however it seems that there is another factor that determines how the images will appear, something like a default
 background lighting.
Is there something that I am missing?
Thank you
[image]
[image]

ArgyChris, 2021-05-14:
It seems that there is an extra lighting source somewhere in the .xml that I missed

alimuldal, 2021-05-16:
It's a bit hard to diagnose without seeing any code. Are you including [`suite/common/visual.xml`](https://github.com/deepmind/dm_control/blob/master/dm_control/suite/common/visual.xml) in your XML? That defines a headlight that could account for the lighting difference.

ArgyChris, 2021-05-17:
Hello, thank you for the reply. The problem was that I didn't include the visual.xml information that, as you said, defines some specifics for the lighting.

closed 2021-05-17 by alimuldal