google-deepmind/dm_control #189: How to change the parameters of mujoco tasks?

ZJP-c4cld, 2021-06-28:
For example, I want to change the length of pole in Cartpole task and observe the effect of this change.

yuvaltassa, 2021-06-28:
You need to tweak the model definition.
For example the length of the pole is defined here
<https://github.com/deepmind/dm_control/blob/master/dm_control/suite/cartpole.xml#L13>
.

Dr. Yuval Tassa | Research Scientist | Google DeepMind | ***@***.***

ZJP-c4cld, 2021-06-28:
@yuvaltassa Thank you for your prompt reply and selfless help ^_^
I've got what you mention. However, I'm not familiar with the relationship between the parameters I want to change and the corresponding definition in XML document. For example, I am unable to find the relationship between the length of the pole and <https://github.com/deepmind/dm_control/blob/master/dm_control/suite/cartpole.xml#L13>...  Because in that code, I don't find words such as "length". How should I learn about the relationship?

yuvaltassa, 2021-06-28:
Read the MuJoCo XML reference <http://mujoco.org/book/XMLreference.html>.
Specifically for your example, read this
<http://mujoco.org/book/XMLreference.html#geom>.
It is very helpful to quickly tweak the XML and reload the model, either in
MuJoCo's simulate app, or in our viewer.

Dr. Yuval Tassa | Research Scientist | Google DeepMind | ***@***.***

ZJP-c4cld, 2021-06-28:
@yuvaltassa Thank you very much.

closed 2021-06-30 by ZJP-c4cld