google-deepmind/dm_control #190: Why state and observation in Control Suite have different dimensions?

ZJP-c4cld, 2021-06-29:
I read the "dm_control: Software and Tasks for Continuous Control". In section 6.2 Domains and Tasks, I notice that state and observation in Control Suite have different dimensions. It makes me a little confused. In my understanding, observation is the partial version of state.(Agents may be unable to global state because of some factors such as fog of war.) The dimension of state should be the same as the one of observation.
[image]

yuvaltassa, 2021-06-29:
The state is the low-level MuJoCo state (position+velocity), observations
are some function of this state.
For example the pendulum and acrobot have 1 and 2 angles (and corresponding
angular velocities) respectively, but in the observation dict the angles
are split into sin/cos
<https://github.com/deepmind/dm_control/blob/master/dm_control/suite/acrobot.py#L111>,
to account for the periodicity.

Dr. Yuval Tassa | Research Scientist | Google DeepMind | ***@***.***

ZJP-c4cld, 2021-06-29:
@yuvaltassa Thank you for your prompt reply and selfless help ^_^

closed 2021-06-29 by ZJP-c4cld