google-deepmind/dm_control #196: loss-function, orientation inverse-kinematics

akssri, 2021-09-06:
The gradient used here is weird,
https://github.com/deepmind/dm_control/blob/master/dm_control/utils/inverse_kinematics.py#L165

```quat2vel``` does axis-angle conversion of the error quaternion ```q q_t^*```, but the jacr here is actually mapping to ```2 vec(\dot{q} q*) = (0, \omega)```. it's unclear what the loss function is, or how the dual (of an undefined loss) is pulled back to id.

vaxenburg, 2021-09-06:
The loss, or rather the objective, is actually given on page 10 in the paper cited here:
https://github.com/deepmind/dm_control/blob/master/dm_control/utils/inverse_kinematics.py#L251

In the code you are referring to, there is no gradient calculation of this objective _per se_. In other words, `err` does not represent the loss in the usual sense and its gradient is not needed. Instead, the objective is minimized directly with respect to `\Delta\theta` given the Jacobian and the error (see Eqs.(10),(11)).

Also, this inverse kinematics problem can be formulated in different ways, including the one you are suggesting, but I think the rotational component of the Jacobian is always required if orientations are considered in the loss.

akssri, 2021-09-07:
I realize that this is using LM for solving non-linear least squares. The issue is that for the residual ```err = \theta vec(q q_t^*)/Z```, currently being used, ```J*err``` is clearly not equal to ```{\partial err \over \partial q}```. It appears that there is no first principle 'objective'+approximation I'm missing, and this is indeed a hack/error.

To reiterate, the Jacobian maps T_q Q -> \mathfrak{so(3)}; this is not the same as {\partial err \over \partial q}.
My only suggestion is to fix this. This can be done by using the correct Jacobian in LM; changing the residual term makes this Jacobian calculation simpler (the axis-angle residual is in any case an approximation of this).

yuvaltassa, 2021-09-08:
Hi @akssri, 

It's not entirely clear to us what you are saying, likely it's a notation thing. In order to avoid further confusion, I'm going to close this comment and as you propose you can send a PR, which should make the discussion much more concrete.

Thanks you so much for your time!

closed 2021-09-08 by yuvaltassa