google-deepmind/dm_control #197: Fast calculation of bounded quaternion distance

vaxenburg, 2021-09-06:
I think there's a faster way of doing this calculation:
https://github.com/deepmind/dm_control/blob/master/dm_control/locomotion/tasks/reference_pose/rewards.py#L26

The bounded distance can be computed with a one-liner `np.arccos(2 * np.sum(quat1 * quat2)**2 - 1)` where `quat1`, `quat2` are (4,) arrays.

I timed about an order of magnitude speedup with this one.

alimuldal, 2021-09-15:
Hi, thanks for the suggestion - do you want to send us a PR?

vaxenburg, 2021-09-15:
Hi @alimuldal, sure, I'll prepare a PR soon

vaxenburg, 2021-09-17:
It doesn't really matter anymore, but another problem with the previous implementation was that is was calling `tr.quat_dist` twice: once to get `default_dist` and once to get `anti_dist`. It's actually sufficient to call `tr.quat_dist` only once to get `default_dist` and then get `anti_dist` simply by `anti_dist = np.pi - default_dist`. This would have been a 2x speedup already.

alimuldal, 2021-09-17:
True, and you could also get the minimum distance using `default_dist % np.pi`. I think there's other low-hanging fruit for optimization in the `transformations` library. Thanks again for contributing!

closed 2021-09-17 by alimuldal

vaxenburg, 2021-09-17:
I'll take a look, maybe I'll spot something to improve there:)