google-deepmind/dm_control #204: Running Simultaneous Agents?

michelhabib, 2021-10-02:
Is it possible to configure DM Control, to run Agents simultaneously? For example, can I have 10 Walkers instead of just one? Is that possible from DM Control side and Mujoco side?
My motivation is doing something like ML-Agents (Unity), which allows running multiple environments at the same time and even multiple unity instances.

alimuldal, 2021-10-04:
Do you mean multiple agents acting within the same simulation, or multiple agents acting in parallel within separate simulations?

For the former you could take a look at the [soccer environments](https://github.com/deepmind/dm_control/tree/master/dm_control/locomotion/soccer) for an example of a multiagent environment implemented in `dm_control`. For the latter you could run multiple copies of your agent-environment interaction loop in separate threads or processes, or perhaps even distributed across multiple machines ([Acme](https://github.com/deepmind/acme) has some relevant examples).

vaxenburg, 2021-10-04:
In `Acme`, a relevant example of running multiple copies of the environment is implemented [here](https://github.com/deepmind/acme/blob/master/acme/agents/tf/d4pg/agent_distributed.py). It leverages [launchpad](https://github.com/deepmind/launchpad) to run the environment copies in parallel in separate threads/processes.