google-deepmind/dm_control #214: libEGL warning: failed to open /dev/dri/renderD128: Permission denied

bitsandscraps, 2021-11-08:
I get the following warnings when running mujoco environments on a headless machine using egl.
I've set the MUJOCO_GL environment variable to egl using the `export MUJOCO_GL=egl` command.

>libEGL warning: failed to open /dev/dri/renderD129: Permission denied
libEGL warning: failed to open /dev/dri/renderD129: Permission denied
libEGL warning: failed to open /dev/dri/renderD128: Permission denied
libEGL warning: failed to open /dev/dri/renderD128: Permission denied

As they are warnings not errors, the program terminates normally, but it feels a bit slow.
#209 seems to have solved a similar problem by using the `nvidia/cudagl` image,
but I'm using a real machine not a virtual one, so the solution cannot be applied directly.

Any help would be greatly appreciated.

saran-t, 2021-11-08:
If you have full control (i.e. root privileges) on this machine, it might simply be a matter of correcting the permissions on `/dev/dri/renderD129` (usually by adding your user to the group that owns the device).

bitsandscraps, 2021-11-08:
> 
> 
> If you have full control (i.e. root privileges) on this machine, it might simply be a matter of correcting the permissions on `/dev/dri/renderD129` (usually by adding your user to the group that owns the device).

After I add myself to the group, the warning message has changed to 
>libEGL warning: DRI2: failed to create dri screen
libEGL warning: DRI2: failed to create dri screen
libEGL warning: DRI2: failed to create dri screen
libEGL warning: DRI2: failed to create dri screen

bitsandscraps, 2021-11-10:
I just checked that even with MUJOCO_GL set to egl, the following test code does not take up any GPU.
```
from dm_control import suite
env = suite.load('cartpole', 'swingup')
for _ in range(100000):
    env.physics.render()
```
The same code utilizes about 60 MiB of GPU memory on my PC that has a physical display.
The graphics driver I installed on my machine is `nvidia-headless-470-server`. Can this be the cause?

saran-t, 2021-12-17:
Sorry for the long silence, been working on getting the MuJoCo 2.1.1 release out of the door. Are you still looking for help on this? If you are, can you give a bit more detail on your system setup?

bitsandscraps, 2021-12-17:
No problem. I've attached a list of packages installed on that machine. What kind of details do you need?
[installed_packages.txt](https://github.com/deepmind/dm_control/files/7737006/installed_packages.txt)

saran-t, 2021-12-17:
It looks like you don't have any Nvidia graphics driver installed at all, only compute ones. I think if you can install https://packages.ubuntu.com/focal/nvidia-driver-470-server that might solve it.

saran-t, 2021-12-17:
Looks like https://packages.ubuntu.com/focal/libnvidia-gl-470-server is actually the minimal thing that's required.

bitsandscraps, 2021-12-18:
`libnvidia-gl-470-server` works like magic! Thanks a lot.

closed 2021-12-18 by bitsandscraps

zzr2002314, 2024-11-26:
libnvidia-gl-470-server works like magic