google-deepmind/dm_control #216: Rope

matansudry, 2021-11-14:
Hi,
How can I remove the golden cube in the default rope of Mujoco?

    <worldbody>
        <body name="B10" pos="0 0 1">
            <freejoint/>
            <composite type="rope" count="21 1 1" spacing="0.04" offset="0 0 2">
                <joint kind="main" damping="0.005"/>
                <geom  type="capsule" size=".01 .015" rgba=".8 .2 .1 1"/>
            </composite>
        </body>
    </worldbody>

[image]

yuvaltassa, 2021-11-14:
Hi,

I loaded this model (after adding opening and closing `<mujoco>` statements) and there is no cube. Clearly you have other things in the scene which you didn't paste in the OP, I suspect one of them is a cube.

closed 2021-11-14 by yuvaltassa

matansudry, 2021-11-15:
Hi Yuval,

I load the pyshich like that,
"
new = "assets/rope_v2.xml"
physics = mujoco.Physics.from_xml_path(new)
"

using those files below and attached, and still having golden cube.

<mujoco model="rope">

<include file="scene_v2.xml"/>

<option timestep="0.002" jacobian="dense"/>

<size nconmax="100" njmax="300" nstack="50000"/>

<worldbody>
<body name="B10" pos="0 -0.5 0.05">
<freejoint/>
<composite type="rope" count="21 1 1" spacing="0.04" offset="0 0 2">
<joint kind="main" damping="0.005"/>
<geom type="capsule" size=".01 .015" rgba=".8 .2 .1 1"/>
</composite>
</body>
</worldbody>
</mujoco>

scene_v2.xml
<mujoco>

<statistic extent="2" meansize=".05"/>

<option timestep="0.005" solver="Newton" iterations="30" tolerance="1e-10"
jacobian="sparse" cone="pyramidal"/>

<visual>
<rgba haze="0.15 0.25 0.35 1"/>
<quality shadowsize="2048"/>
<map stiffness="700" shadowscale="0.5" fogstart="10" fogend="15" zfar="40"
haze="0.3"/>
</visual>

<asset>
<texture type="skybox" builtin="gradient" rgb1="0.3 0.5 0.7" rgb2="0 0 0"
width="512" height="512"/>
<texture name="texplane" type="2d" builtin="checker" rgb1=".2 .3 .4" rgb2=".1
0.15 0.2"
width="512" height="512" mark="cross" markrgb=".8 .8 .8"/>
<texture name="texcarpet" type="2d" file="carpet.png"/>
<texture name="texsponge" type="2d" file="sponge.png"/>
<texture name="texmarble" type="cube" file="marble.png"/>

<material name="matplane" reflectance="0.3" texture="texplane" texrepeat="1
1" texuniform="true"/>
<material name="matcarpet" texture="texcarpet"/>
<material name="matsponge" texture="texsponge" specular="0.3"/>
<material name="matmarble" texture="texmarble" rgba=".7 .7 .7 1"/>
</asset>

<worldbody>
<light directional="true" diffuse=".4 .4 .4" specular="0.1 0.1 0.1" pos="0
0 5.0" dir="0 0 -1" castshadow="false"/>
<light directional="true" diffuse=".6 .6 .6" specular="0.2 0.2 0.2" pos="0
0 4" dir="0 0 -1"/>

<geom name="ground" type="plane" size="0 0 1" pos="0 0 0" quat="1 0 0 0"
material="matplane" condim="1"/>

</worldbody>
</mujoco>

yuvaltassa, 2021-11-15:
Hi,
Note it is possible to attach files to github comments (drag and drop). Could you please attach a zip containing a loadable model (including e.g. assets) that we could inspect?
Thanks

matansudry, 2021-11-15:
attached are the bug script
[rope_bug.zip](https://github.com/deepmind/dm_control/files/7539292/rope_bug.zip)

yuvaltassa, 2021-11-15:
Hi,
The box you are seeing is simply the visualisation of the free joint at the base body, which you are enabling in your script with
```
scene_option = mujoco.wrapper.core.MjvOption()
scene_option.flags[enums.mjtVisFlag.mjVIS_JOINT] = True
```
Delete the `scene_option` struct (it is not required) and the box will go away. What threw me off initially is the colours. The default for joints is blue, not yellow, but then your capsules are supposed to be red and come out blue, so I'm not sure what is happening with your rendering. Below is what your model looks like when I load it in `simulate` and also activate visualisation for the rope joints (they are hidden by default (group 3), to enable their visualisation add `group=0` to the `<joint>` specifier, or click the `joint 3` button under "group enable" in `simulate`).

[image]

saran-t, 2021-11-15:
The colour flip is due to OpenCV interpreting the pixel array as BGR rather than RGB.