google-deepmind/dm_control #220: dm_control.rl.control.PhysicsError: Physics state is invalid. Warning(s) raised: mjWARN_CONTACTFULL

13253591602, 2021-11-30:
Hi, I am trying to reproduce the experiment from paper "Learning human behaviors from motion capture by adversarial imitation". I used the code from https://github.com/huiwenzhang/merel-mocap-gail. But after running about 400 iterations, it throws the following error:
WARNING:absl:Pre-allocated contact buffer is full. Increase nconmax above 100. Time = 0.0000.
Traceback (most recent call last):
  File "/home/yh/anaconda3/envs/gail/lib/python3.7/runpy.py", line 193, in _run_module_as_main
    "__main__", mod_spec)
  File "/home/yh/anaconda3/envs/gail/lib/python3.7/runpy.py", line 85, in _run_code
    exec(code, run_globals)
  File "/home/yh/Projects/IL/merel-mocap-gail-master/baselines/baselines/gail/run_mujoco.py", line 268, in <module>
    main(args)
  File "/home/yh/Projects/IL/merel-mocap-gail-master/baselines/baselines/gail/run_mujoco.py", line 125, in main
    args.obs_only
  File "/home/yh/Projects/IL/merel-mocap-gail-master/baselines/baselines/gail/run_mujoco.py", line 181, in train
    obs_only=obs_only)
  File "/home/yh/Projects/IL/merel-mocap-gail-master/baselines/baselines/gail/trpo_mpi.py", line 316, in learn
    seg = seg_gen.__next__()
  File "/home/yh/Projects/IL/merel-mocap-gail-master/baselines/baselines/gail/trpo_mpi.py", line 132, in traj_segment_generator
    env.reset()
  File "/home/yh/anaconda3/envs/gail/lib/python3.7/site-packages/dm_control/rl/control.py", line 82, in reset
    self._task.initialize_episode(self._physics)
  File "/home/yh/anaconda3/envs/gail/lib/python3.7/site-packages/dm_control/suite/humanoid_CMU.py", line 136, in initialize_episode
    physics.after_reset()
  File "/home/yh/anaconda3/envs/gail/lib/python3.7/site-packages/dm_control/mujoco/engine.py", line 298, in after_reset
    self.forward()
  File "/home/yh/anaconda3/envs/gail/lib/python3.7/site-packages/dm_control/mujoco/engine.py", line 308, in forward
    mjlib.mj_forward(self.model.ptr, self.data.ptr)
  File "/home/yh/anaconda3/envs/gail/lib/python3.7/contextlib.py", line 119, in __exit__
    next(self.gen)
  File "/home/yh/anaconda3/envs/gail/lib/python3.7/site-packages/dm_control/mujoco/engine.py", line 331, in check_invalid_state
    raise _control.PhysicsError(message)
dm_control.rl.control.PhysicsError: Physics state is invalid. Warning(s) raised: mjWARN_CONTACTFULL

I have tried all dm_control version.

Markus28, 2021-11-30:
Somewhat connected to https://github.com/deepmind/deepmind-research/issues/307

vaxenburg, 2022-02-07:
This happens when the number of detected contacts is larger than the allocated contact buffer (as the error message says). It is explained, e.g., [here](https://mujoco.readthedocs.io/en/latest/XMLreference.html#size). One way to increase the buffer size is to tweak the `nconmax` parameter in your model's XML file. You want to look for an entry that might look like `<size nconmax="100"/>`.