google-deepmind/dm_control #223: Trying out M1 MacBook...

slerman12, 2021-12-17:
I installed the latest release 2.1.1, and made sure to run 'brew install glow'. Then I updated dm_control with pip... however I am still getting this error:

'Unable to load EGL library', 'dlopen(EGL, 10): image not found', 'EGL', None

I'm not sure what the problem is. My code sets env variables as follows:

```
MKL_SERVICE_FORCE_INTEL: '1'
MUJOCO_GL: 'egl'
```

But the problem persists eve when I get rid of those.

saran-t, 2021-12-17:
For macOS you should set MUJOCO_GL=glfw. You'll also need to `pip install glfw` first.

slerman12, 2021-12-17:
Tried:

```
(base) samlerman@Marvin .mujoco % pip install glfw
Requirement already satisfied: glfw in /Users/samlerman/code/miniforge3/lib/python3.9/site-packages (2.3.0)
```

And:

```
MKL_SERVICE_FORCE_INTEL: '1'
MUJOCO_GL: 'glfw'
```

as well as:

`MUJOCO_GL: 'glfw'`

But I get a new error now:

`_default_error_callback = _GLFWerrorfun(_handle_glfw_errors)`

saran-t, 2021-12-17:
Can you paste the full error? I don't have enough context from that to see what's going on.

slerman12, 2021-12-17:
Sorry, this is what I have:

  ```
File "/Users/samlerman/Code/U/DMC.py", line 10, in make
    from dm_control import manipulation, suite
  File "/Users/samlerman/Code/miniforge3/lib/python3.9/site-packages/dm_control/manipulation/__init__.py", line 20, in <module>
    from dm_control import composer as _composer
  File "/Users/samlerman/Code/miniforge3/lib/python3.9/site-packages/dm_control/composer/__init__.py", line 18, in <module>
    from dm_control.composer.arena import Arena
  File "/Users/samlerman/Code/miniforge3/lib/python3.9/site-packages/dm_control/composer/arena.py", line 20, in <module>
    from dm_control import mjcf
  File "/Users/samlerman/Code/miniforge3/lib/python3.9/site-packages/dm_control/mjcf/__init__.py", line 18, in <module>
    from dm_control.mjcf.attribute import Asset
  File "/Users/samlerman/Code/miniforge3/lib/python3.9/site-packages/dm_control/mjcf/attribute.py", line 28, in <module>
    from dm_control.mujoco.wrapper import util
  File "/Users/samlerman/Code/miniforge3/lib/python3.9/site-packages/dm_control/mujoco/__init__.py", line 18, in <module>
    from dm_control.mujoco.engine import action_spec
  File "/Users/samlerman/Code/miniforge3/lib/python3.9/site-packages/dm_control/mujoco/engine.py", line 41, in <module>
    from dm_control import _render
  File "/Users/samlerman/Code/miniforge3/lib/python3.9/site-packages/dm_control/_render/__init__.py", line 67, in <module>
    Renderer = import_func()  # pylint: disable=invalid-name
  File "/Users/samlerman/Code/miniforge3/lib/python3.9/site-packages/dm_control/_render/__init__.py", line 41, in _import_glfw
    from dm_control._render.glfw_renderer import GLFWContext
  File "/Users/samlerman/Code/miniforge3/lib/python3.9/site-packages/dm_control/_render/glfw_renderer.py", line 24, in <module>
    import glfw  # pylint: disable=g-import-not-at-top
  File "/Users/samlerman/Code/miniforge3/lib/python3.9/site-packages/glfw/__init__.py", line 842, in <module>
    _default_error_callback = _GLFWerrorfun(_handle_glfw_errors)
MemoryError: Error instantiating 'Datasets.Environments.Environment.Environment' : 
```

slerman12, 2021-12-17:
(MemoryError is returned by a context manager)

saran-t, 2021-12-17:
Can you try to see if you get an error by just firing up `python3.9` and typing in `import glfw` ?

slerman12, 2021-12-17:
Yes! Same thing, even after reinstalling!

saran-t, 2021-12-17:
It looks like you're trying to run this within Conda, which isn't something that we've tested. Can you try running inside a vanilla Python virtualenv?

slerman12, 2021-12-17:
There are a lot of packages to install... but I think it should work since I just created a vanilla virtualenv and ran `import glfw` (after installing glfw) with no problem.

saran-t, 2021-12-17:
May be look around online to see if there are any reported issues with GLFW in Conda?

saran-t, 2021-12-17:
Just a hunch: can you run `find /Users/samlerman/Code/miniforge3 -iname \*glfw\*dylib\*` for me?

slerman12, 2021-12-17:
Hmm, here's the output:

```
/Users/samlerman/Code/miniforge3/pkgs/glfw-3.3.6-h3422bc3_0/lib/libglfw.dylib
/Users/samlerman/Code/miniforge3/pkgs/glfw-3.3.6-h3422bc3_0/lib/libglfw.3.dylib
/Users/samlerman/Code/miniforge3/pkgs/glfw-3.3.6-h3422bc3_0/lib/libglfw.3.3.dylib
/Users/samlerman/Code/miniforge3/pkgs/glfw-3.3.4-h3422bc3_0/lib/libglfw.dylib
/Users/samlerman/Code/miniforge3/pkgs/glfw-3.3.4-h3422bc3_0/lib/libglfw.3.dylib
/Users/samlerman/Code/miniforge3/pkgs/glfw-3.3.4-h3422bc3_0/lib/libglfw.3.3.dylib
/Users/samlerman/Code/miniforge3/lib/libglfw.dylib
/Users/samlerman/Code/miniforge3/lib/libglfw.3.dylib
/Users/samlerman/Code/miniforge3/lib/libglfw.3.3.dylib
```

saran-t, 2021-12-17:
Can you prepend `DYLD_LIBRARY_PATH="/Users/samlerman/Code/miniforge3/lib:${DYLD_LIBRARY_PATH}` in front if the command that you used to launch your Python script?

slerman12, 2021-12-17:
Sure, I'm using Hydra to manage args, so that seems to be the crux of the output:

```
/Run.py: line 5: import: command not found
from: can't read /var/mail/hydra.utils
/Run.py: line 8: import: command not found
from: can't read /var/mail/pathlib
/Run.py: line 11: import: command not found
/Run.py: line 13: import: command not found
from: can't read /var/mail/torch.backends
/Run.py: line 16: cudnn.benchmark: command not found
/Run.py: line 22: syntax error near unexpected token `config_path='Hyperparams
```

The problem seems to be with glfw though...

saran-t, 2021-12-17:
Library lookup paths for pyGLFW is defined [here](https://github.com/FlorianRhiem/pyGLFW/blob/master/glfw/library.py#L129). According to [their PyPI page], they say that `the GLFW shared library for 64-bit is included in the Python wheels for macOS`, however according to your `find` output it seems that for whatever reason this isn't the case in Conda. One would expect that Conda would set `sys.prefix` to `/Users/samlerman/Code/miniforge3` for you, but it seems that's not the case. I think you easiest option would be to install [Homebrew](https://brew.sh), then `brew install glfw`.

slerman12, 2021-12-17:
Unfortunately, that's the first thing I tried:

```
% brew install glfw

Warning: glfw 3.3.6 is already installed and up-to-date.
```

saran-t, 2021-12-17:
Can you paste the full stack trace when you just manually run `import glfw` in the Python REPL?

slerman12, 2021-12-17:
After uninstalling the pip version:

```
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
ModuleNotFoundError: No module named 'glfw'
```

This was with the conda version.

Otherwise, the pip version throws:

```
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/Users/samlerman/code/miniforge3/lib/python3.9/site-packages/glfw/__init__.py", line 914, in <module>
    _default_error_callback = _GLFWerrorfun(_handle_glfw_errors)
MemoryError
```

saran-t, 2021-12-17:
Wait, have you installed `pyGLFW` (as opposed to just GLFW) through Conda? `No module named 'glfw'` suggests that pyGLFW isn't installed.

slerman12, 2021-12-17:
No, should I give that a shot? I may have tried it back before 2.1.1.

saran-t, 2021-12-17:
Yeah, install https://anaconda.org/conda-forge/pyglfw

slerman12, 2021-12-17:
Still: :/

```
>>> import glfw
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/Users/samlerman/code/miniforge3/lib/python3.9/site-packages/glfw/__init__.py", line 914, in <module>
    _default_error_callback = _GLFWerrorfun(_handle_glfw_errors)
MemoryError
```

slerman12, 2022-02-01:
Any update on this?

closed 2022-02-01 by slerman12