google-deepmind/dm_control #224: Prevent potential collisions with environment in custom manipulation tasks

xf-zhao, 2021-12-24:
Hello, thank you all for this great work! This acctually has nothing to do with a bug. I just have a question regarding to `manipulation` environment, which confused me a lot (even after spend lots of time reading the source code of dmc): 

Who (Mujoco or dmc?) or which part of the code (if it is dmc) defines the "never touch the floor" behaviour of a Jaco arm? I want to implement similar thing in a customed env and (in the future) on a real robot arm. This behaviour is quite attractive because it could prevent the arm do any harm (to desk or people if set them as dynamic `Entity` I think).

Code to demonstrate the "never touch floor" behaviour (which I want to know how it is defined or maybe how to modify for personal preference).

``` python
from dm_control import manipulation
from dm_control import viewer
import numpy as np

env = manipulation.load('reassemble_3_bricks_fixed_order_vision', seed=42)
action_spec = env.action_spec()

# Define a uniform random policy.
def touch_ground_policy(time_step):
  del time_step  # Unused.
  a = np.random.uniform(low=action_spec.minimum,
                           high=action_spec.maximum,
                           size=action_spec.shape)
  a = np.zeros_like(a)
  a[1] = -100
  return a

# Launch the viewer application.
viewer.launch(env, policy=touch_ground_policy)
```

To be honest I not familiar with Mujoco and I would appreciate it a lot if you can help me with this purpose. Thank you very much!

alimuldal, 2022-01-03:
I am a bit confused by your question. It sounds as though you are looking for some sort of safety mechanism that would prevent the arm from colliding with the floor, but no such mechanism exists for the manipulation tasks, and if I run your example script the arm *does* collide with the floor:

[image]

The hand doesn't always hit the floor because the orientation of the joint at the base of the arm varies between trials, so sometimes it "folds backwards" at the elbow, e.g.:

[image]

In this GIF I made the geoms translucent ("T") and enabled the arrows representing contact forces ("F") so you can see the contact forces acting on the arm.

It's definitely possible to design a safety mechanism like you describe, although that's not something we intend to provide with `dm_control`.

xf-zhao, 2022-04-03:
Thank you for your answer. I figured out finally. Your design of manipulation tasks is very stable, particularly useful for RL training. I learned a lot from the codes. Thank you again for the work.

closed 2022-04-03 by xf-zhao