google-deepmind/dm_control #226: OSError: /root/.mujoco/mujoco-2.1.1/lib/libmujoco.so.2.1.1: undefined symbol: __glewActiveTexture

LSC527, 2021-12-30:
platform: ubuntu18.04
mujoco version: 2.1.1
```
  File "/root/miniconda3/envs/lowprec_rl/lib/python3.7/site-packages/dm_control/suite/__init__.py", line 24, in <module>
    from dm_control.suite import acrobot
  File "/root/miniconda3/envs/lowprec_rl/lib/python3.7/site-packages/dm_control/suite/acrobot.py", line 20, in <module>
    from dm_control import mujoco
  File "/root/miniconda3/envs/lowprec_rl/lib/python3.7/site-packages/dm_control/mujoco/__init__.py", line 18, in <module>
    from dm_control.mujoco.engine import action_spec
  File "/root/miniconda3/envs/lowprec_rl/lib/python3.7/site-packages/dm_control/mujoco/engine.py", line 42, in <module>
    from dm_control.mujoco import index
  File "/root/miniconda3/envs/lowprec_rl/lib/python3.7/site-packages/dm_control/mujoco/index.py", line 88, in <module>
    from dm_control.mujoco.wrapper import util
  File "/root/miniconda3/envs/lowprec_rl/lib/python3.7/site-packages/dm_control/mujoco/wrapper/__init__.py", line 20, in <module>
    from dm_control.mujoco.wrapper.core import callback_context
  File "/root/miniconda3/envs/lowprec_rl/lib/python3.7/site-packages/dm_control/mujoco/wrapper/core.py", line 29, in <module>
    from dm_control.mujoco.wrapper.mjbindings import functions
  File "/root/miniconda3/envs/lowprec_rl/lib/python3.7/site-packages/dm_control/mujoco/wrapper/mjbindings/functions.py", line 17, in <module>
    mjlib = util.get_mjlib()
  File "/root/miniconda3/envs/lowprec_rl/lib/python3.7/site-packages/dm_control/mujoco/wrapper/util.py", line 123, in get_mjlib
    return ctypes.cdll.LoadLibrary(library_path)
  File "/root/miniconda3/envs/lowprec_rl/lib/python3.7/ctypes/__init__.py", line 442, in LoadLibrary
    return self._dlltype(name)
  File "/root/miniconda3/envs/lowprec_rl/lib/python3.7/ctypes/__init__.py", line 364, in __init__
    self._handle = _dlopen(self._name, mode)
OSError: /root/.mujoco/mujoco-2.1.1/lib/libmujoco.so.2.1.1: undefined symbol: __glewActiveTexture
```

saran-t, 2022-01-04:
Do you need headless rendering? If not, I think this can be resolved by following the instructions in the README and `sudo apt-get install libglfw3 libglew2.0`

LSC527, 2022-01-05:
Thanks, solved.

closed 2022-01-05 by LSC527