google-deepmind/dm_control #233: Mujoco library error in dm_control

plaban, 2022-01-13:
Hi,
While configuring dm_control with mushroom_rl, I am finding difficulty in configuring mujoco200. Following is the error:

AttributeError: <HOME>/.mujoco/mujoco200/bin/libmujoco200.so: undefined symbol: mjr_label

Following are the environment variables set:
export MUJOCO_GL=osmesa
export MJLIB_PATH="<HOME>/.mujoco/mujoco200/bin/libmujoco200.so"
export MJKEY_PATH="<HOME>/.mujoco/mujoco200/mjkey.txt"
export MUJOCO_PY_MJPRO_PATH="<HOME>/.mujoco/mujoco200/"
export MUJOCO_PY_MJKEY_PATH="<HOME>/.mujoco/mujoco200/mjkey.txt"
export LD_LIBRARY_PATH="$LD_LIBRARY_PATH:<HOME>/.mujoco/mujoco200/bin"

I am stuck for a long at this point. Kindly help

qizhg, 2022-01-24:
Try switching to mujoco210.

saran-t, 2022-01-31:
As the above comment suggested, please try the latest version of dm_control with MuJoCo 2.1.1.

saran-t, 2022-03-15:
As of today, if you just `pip install dm_control` in a fresh virtualenv you should get version 1.0.0, which automatically pulls in a copy of MuJoCo so that you no longer need to copy things around yourself.

closed 2022-03-15 by saran-t