google-deepmind/dm_control #234: pip install error

slerman12, 2022-01-14:
```
ERROR: Command errored out with exit status 1:
     command: /u/slerman/miniconda3/envs/L/bin/python -u -c 'import io, os, sys, setuptools, tokenize; sys.argv[0] = '"'"'/tmp/pip-req-build-21crwfji/setup.py'"'"'; __file__='"'"'/tmp/pip-req-build-21crwfji/setup.py'"'"';f = getattr(tokenize, '"'"'open'"'"', open)(__file__) if os.path.exists(__file__) else io.StringIO('"'"'from setuptools import setup; setup()'"'"');code = f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();exec(compile(code, __file__, '"'"'exec'"'"'))' install --record /tmp/pip-record-saom06fy/install-record.txt --single-version-externally-managed --compile --install-headers /u/slerman/miniconda3/envs/L/include/python3.8/dm-control
         cwd: /tmp/pip-req-build-21crwfji/
    Complete output (2 lines):
    running install
    error: Header file 'mjdata.h' does not exist.
    ----------------------------------------
ERROR: Command errored out with exit status 1: /u/slerman/miniconda3/envs/L/bin/python -u -c 'import io, os, sys, setuptools, tokenize; sys.argv[0] = '"'"'/tmp/pip-req-build-21crwfji/setup.py'"'"'; __file__='"'"'/tmp/pip-req-build-21crwfji/setup.py'"'"';f = getattr(tokenize, '"'"'open'"'"', open)(__file__) if os.path.exists(__file__) else io.StringIO('"'"'from setuptools import setup; setup()'"'"');code = f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();exec(compile(code, __file__, '"'"'exec'"'"'))' install --record /tmp/pip-record-saom06fy/install-record.txt --single-version-externally-managed --compile --install-headers /u/slerman/miniconda3/envs/RL/include/python3.8/dm-control Check the logs for full command output.
```

HzcIrving, 2022-01-18:
How to fix it ?

jackyoung96, 2022-01-21:
Same problem

jackyoung96, 2022-01-24:
I used Python 3.7, 3.8, and mujoco200. Then I got the same error.

I solved it by using Python 3.6 and mujoco 210.
One more thing is that modifying the folder name of mujoco
from
~/.mujoco/mujoco210 
to 
~/.mujoco/mujoco200_linux

aleSuglia, 2022-01-25:
Same here. Installation seems broken...

aleSuglia, 2022-01-25:
Also, I wanted to mention that there are other errors in the references to the mujoco libraries when the code is actually executed. So I've resolved by making sure that inside `~/.mujoco/lib` I have the following files:

```
libglewegl.so  libglewosmesa.so  libglew.so  libglfw3.a  libglfw.so.3  libmujoco210nogl.so  libmujoco210.so  libmujoco.so.2.1.1
```
where `libmujoco.so.2.1.1` is just a symlink to `libmujoco210.so`.

saran-t, 2022-01-31:
Sorry, everyone. There may have been a bit of confusion since for a week or so we've updated our GitHub repo but didn't push a new wheel to PyPI.

As things stand now, you should download MuJoCo 2.1.1 and
- On Linux, extract the entire tarball into `~/.mujoco`
- On macOS, place `MuJoCo.app` into `/Applications`
- On Windows, extract the entire zip archive into `%HOME%\MuJoCo`

Then you should be able to simply `pip install dm_control` and everything should just work, no symlink/renaming required. Please let me know if that's not the case.

closed 2022-01-31 by saran-t