google-deepmind/dm_control #235: After installation I am not being able to import dm_control suit on google colaboratory. Could you please help?

navidyou, 2022-01-16:
After successfully installing dm_control and Mujoco on google colaboratory. It is not possible to import dm_control suit on google colaboratory. The following code gives the following error:

from dm_control import suite
.
.
Cannot find MuJoCo library at /root/.mujoco/mujoco210/bin/libmujoco210.so

Although I have successfully installed mujuco210 on google colab and "import mujoco_py" is executed without any errors. 

Could you please help fix this so we can use dm_control on google colab? This would be so valuable.

Thank you so much,

navidyou, 2022-01-16:
#the following is the code for installing mujoco on google colab. The library path should be '/root/.mujoco/mujoco210/bin' without any issues! 

import os
if not os.path.exists('.mujoco_setup_complete'):
  # Get the prereqs
  !apt-get -qq update
  !apt-get -qq install -y libosmesa6-dev libgl1-mesa-glx libglfw3 libgl1-mesa-dev libglew-dev patchelf
  # Get Mujoco
  !mkdir ~/.mujoco
  !wget -q https://mujoco.org/download/mujoco210-linux-x86_64.tar.gz -O mujoco.tar.gz
  !tar -zxf mujoco.tar.gz -C "$HOME/.mujoco"
  !rm mujoco.tar.gz
  # Add it to the actively loaded path and the bashrc path (these only do so much)
  !echo 'export LD_LIBRARY_PATH=$LD_LIBRARY_PATH:$HOME/.mujoco/mujoco210/bin' >> ~/.bashrc 
  !echo 'export LD_PRELOAD=$LD_PRELOAD:/usr/lib/x86_64-linux-gnu/libGLEW.so' >> ~/.bashrc 
  # THE ANNOYING ONE, FORCE IT INTO LDCONFIG SO WE ACTUALLY GET ACCESS TO IT THIS SESSION
  !echo "/root/.mujoco/mujoco210/bin" > /etc/ld.so.conf.d/mujoco_ld_lib_path.conf
  !ldconfig
  # Install Mujoco-py
  !pip3 install -U 'mujoco-py<2.2,>=2.1'
  # run once
  !touch .mujoco_setup_complete

try:
  if _mujoco_run_once:
    pass
except NameError:
  _mujoco_run_once = False
if not _mujoco_run_once:
  # Add it to the actively loaded path and the bashrc path (these only do so much)
  try:
    os.environ['LD_LIBRARY_PATH']=os.environ['LD_LIBRARY_PATH'] + ':/root/.mujoco/mujoco210/bin'
  except KeyError:
    os.environ['LD_LIBRARY_PATH']='/root/.mujoco/mujoco210/bin'
  try:
    os.environ['LD_PRELOAD']=os.environ['LD_PRELOAD'] + ':/usr/lib/x86_64-linux-gnu/libGLEW.so'
  except KeyError:
    os.environ['LD_PRELOAD']='/usr/lib/x86_64-linux-gnu/libGLEW.so'
  # presetup so we don't see output on first env initialization
  import mujoco_py
  _mujoco_run_once = True

saran-t, 2022-01-16:
This seems to be a mujoco_py issue rather than a dm_control one?

alimuldal, 2022-01-17:
I am not sure why you are manipulating `LD_LIBRARY_PATH` and `LD_PRELOAD` - this should not be necessary with `dm_control` (perhaps it's `mujoco_py`-related?). The [tutorial notebook](https://colab.research.google.com/github/deepmind/dm_control/blob/master/tutorial.ipynb) contains a working example of how to install `dm_control` on colab, so I suggest you use that as a starting point.

navidyou, 2022-01-21:
@saran-t The problem was with my dm_control installation. I was able to properly install dm_control by following the tutorial notebook https://colab.research.google.com/github/deepmind/dm_control/blob/master/tutorial.ipynb

navidyou, 2022-01-21:
@alimuldal alimuldal Thank you so much, 

I followed the tutorial in the link you provided and was able to properly install dm_control

 https://colab.research.google.com/github/deepmind/dm_control/blob/master/tutorial.ipynb

closed 2022-01-21 by navidyou