google-deepmind/dm_control #236: Discount value should be None, not 1.0 for case of non-terminal step

ElliotMunro200, 2022-01-17:
https://github.com/deepmind/dm_control/blob/2346e9f6c8fb15e6ea8d8f8bffd0a23854e650ce/dm_control/rl/control.py#L123

alimuldal, 2022-01-17:
Why do you think it should be `None`?

https://github.com/deepmind/dm_env/blob/master/docs/index.md#example-sequences

ElliotMunro200, 2022-01-18:
My bad, it shouldn't be None. I was confused about the values coming through to the discount variable from .get_termination() on line 114. The value is None if the episode shouldn't terminate, but not None if it should terminate. I wrongly assumed that the discount value for StepType.MID to be returned was likewise None. Thanks for posting the link it was very helpful.

closed 2022-01-18 by ElliotMunro200