google-deepmind/dm_control #239: rendering not working on macOS for version 2.1.1

enajx, 2022-01-19:
Following the readme steps (pip install from last commit) the library can be install, however it can not render despite glfw being installed and the library path being set. Running the sample script:

```
from dm_control import suite
from dm_control import viewer

env = suite.load(domain_name="humanoid", task_name="stand")
viewer.launch(env)
```

produces the following error:

```
dm.py 27 <module>
viewer.launch(env)

__init__.py 39 launch
app = application.Application(title=title, width=width, height=height)

application.py 187 __init__
self._window = gui.RenderWindow(width, height, title)

__init__.py 34 ErrorRenderWindow
raise ImportError(

ImportError:
Cannot create a window because no windowing system could be imported
```

Using python 3.9, intel-based Mac.

saran-t, 2022-01-31:
Thanks for reporting. This appears to be caused by my commit 73c31ac21d1214f0a2eadfc8540a7f76796955cc rather than a breakage in MuJoCo 2.1.1. I'll work on reverting the offending commit now.

closed 2022-01-31 by saran-t

kevinzakka, 2022-01-31:
Thanks @saran-t , this seems to have fixed rendering GLFW issues on my mac :-)!

enajx, 2022-03-26:
@saran-t I'm still getting the same initial issue with the new mujoco installation via pip 😢

commit fe444960 (2022-01-31): Revert commit 73c31ac21d1214f0a2eadfc8540a7f76796955cc.

That commit broke rendering on macOS. `glfwInitHint` only sets internal hints in GLFW and a subsequent call to `glfwInit` is still required. However, `glfwInitHint(GLFW_COCOA_MENUBAR, 0); glfwInit();` still results in a dock icon being created, so we are reverting that commit entirely until we figure out a way to actually disable it.

Closes #239

PiperOrigin-RevId: 425341097
Change-Id: If87643db372d4ec0a5237deb6c0b3effecdd7dc0