google-deepmind/dm_control #241: Creating a Ball with PyMJCF

ishank-juneja, 2022-01-24:
I am new to both `dm_control` and mujoco and really liked the idea of creating MJCF models programatically without using `.xml` files, so I was trying to create a model of an un-actuated ball, freely falling under `g` using `dm_control.composer`.

This is what I came up with initially,
```
class Ball(composer.Entity):
    """A Ball derived from composer.Entity base class"""
    def _build(self, radius, mass):
        self._model = self.make_MJCF_ball(radius, mass)

    def _build_observables(self):
        return BallObservables(self)

    @property
    def mjcf_model(self):
        return self._model

    # Create a ball in MJCF using dm_control's PyMJCF API
    def make_MJCF_ball(self, radius, mass):
        # Pick a random opaque color for the ball
        rgba = np.random.uniform([0, 0, 0, 1], [1, 1, 1, 1])
        model = mjcf.RootElement()
        # Make the sphere geom.
        sphere = model.worldbody.add('geom', name='sphere', type='sphere', size=0.1, rgba=rgba)
        return model

class BallObservables(composer.Observables):
    """Observable quantities for the ball/freely falling sphere."""
    @composer.observable
    def ball_position(self):
        sphere_geom = self._entity.mjcf_model.find_all('geom')
        return observable.MJCFFeature('qpos', sphere_geom)

    @composer.observable
    def ball_velocity(self):
        sphere_geom = self._entity.mjcf_model.find_all('geom')
        return observable.MJCFFeature('qvel', sphere_geom)

```

I based my sphere creation off of the below line from your tutorial,
`# Make the torso geom.model.worldbody.add('geom', name='torso', type='ellipsoid', size=BODY_SIZE, rgba=rgba)` 
and the fact that the `.xml` file for this sphere would include the line, `<geom name="sphere" pos=".2 .2 .2" size=".1" rgba="1 1 1 1"/>`,

However when I attempt to create `Ball(1, 1)` (second param mass is unused for now), I get the error,<br>
`IndexError: during initialization of attribute 'size' of element <geom>: tuple index out of range`

This error goes away if I get rid of the `size=` parameter but that prevents me from setting the size of the sphere. I suspect I shouldn't be using `model.worldbody.add` but something else but I am not sure what since I couldn't find more examples online.

vaxenburg, 2022-01-24:
This should work I think:
`sphere = model.worldbody.add('geom', name='sphere', type='sphere', size=(radius,), rgba=rgba)`

ishank-juneja, 2022-01-24:
Thanks @vaxenburg, closing Issue

closed 2022-01-24 by ishank-juneja