google-deepmind/dm_control #243: Help needed with using `dm_control.composer`

ishank-juneja, 2022-01-25:
I really like the `dm_control.composer` interface for creating MJCF models with python. To understand its usage better I am trying to create a model for an isolated ball freely falling under `gravity=(0., 0., 9.8)` starting with some initial velocity from an arbitrary point within the cam FOV. I am new to both `dm_control` and MuJoCo and so far I have been able to create a geometrical ball and add it to the default `Arena`.

```
from dm_control import composer
from dm_control import mjcf
from dm_control.composer.observation import observable  # For exposing observations
from dm_control.composer import variation   # Implements model variations
from dm_control.composer import Arena   # Blank canvas arena
import numpy as np

NUM_SUBSTEPS = 25  # The number of physics sub-steps per control time step.

class Ball(composer.Entity):
    """A Ball derived from composer.Entity base class"""
    def _build(self, radius, mass):
        # Create a mjcf root associated with this entity
        self._mjcf_model = mjcf.RootElement()
        # Pick a random opaque color for the ball
        rgba = np.random.uniform([0, 0, 0, 1], [1, 1, 1, 1])
        # Since this entity is a single object geometry, we can add that sphere geometry and assign as _geom
        self._geom = self._mjcf_model.worldbody.add('geom', name='sphere', type='sphere', size=[radius], rgba=rgba)

    def _build_observables(self):
        return BallObservables(self)

    @property
    def mjcf_model(self):
        return self._mjcf_model

    # Unused actuators property for balls
    @property
    def actuators(self):
        return tuple(self._mjcf_model.find_all('actuator'))

class BallObservables(composer.Observables):
    """Observable quantities for the ball/freely falling sphere."""
    @composer.observable
    def ball_position(self):
        # Return ptrs to all geoms, just a sphere in case of ball
        sphere_geom = self._entity.mjcf_model.find_all('geom')
        return observable.MJCFFeature('pos', sphere_geom)

    @composer.observable
    def ball_velocity(self):
        # Return ptrs to all geoms, just a sphere in case of ball
        sphere_geom = self._entity.mjcf_model.find_all('geom')
        return observable.MJCFFeature('vel', sphere_geom)

# The task the ball has to complete is to fall under gravity and fall out of view
# A composer.Task object combines all the Entities (just Ball here) and is used with a composer.Environment Wrapper
class FallOutOfView(composer.Task):
    # Set up the scene
    def __init__(self, ball):
        # Create an Arena for the task using default blank Arena
        self._arena = Arena()
        # Add the ball object to the Arena
        self._ball = ball
        self._arena.add_free_entity(self._ball)
        # Configure lighting
        self._arena.mjcf_model.worldbody.add('light', pos=(0, 0, 4))

        # Configure initial pose
        self._ball_initial_pose = (0., 0., 0.)

        # Configure variators
        # For varying MJCF attributes like geom size
        # Called before initializing an episode before the model is compiled
        self._mjcf_variator = variation.MJCFVariator()
        # Changes bound attributes like external forces
        # Called after model has been compiled
        self._physics_variator = variation.PhysicsVariator()

        # Enable observables
        self._ball.observables.ball_position.enabled = True
        # self._ball.observables.ball_velocity.enabled = True

        # NUM_SUBSTEPS physics steps per control time step
        self.control_timestep = NUM_SUBSTEPS * self.physics_timestep
        self._task_observables = {}

    #  - - - - call_backs - - - - -

    @property
    def root_entity(self):
        return self._arena

    @property
    def task_observables(self):
        return self._task_observables

    def initialize_episode_mjcf(self, random_state):
        self._mjcf_variator.apply_variations(random_state)

    def initialize_episode(self, physics, random_state):
        self._physics_variator.apply_variations(physics, random_state)

    def get_reward(self, physics):
        return 0.0
```     

Currently the task `FallOutOfView(composer.Task)` that my `env` will wrap around is a dummy `composer.Task` object that doesn't do anything. To be able to achieve its intended task: which is to let the ball fall under gravity starting from some initial velocity untill it is out of view, I need to add velocity, and gravity into the picture which I am not sure how I can/should do.

## My Questions
1. If I were creating a `ball.xml` MJCF, then to add gravity, I would have added the below line, the equivalent of which I unable to find in the `dm_control.composer` tutorial.
```
<option gravity="1e-5 0 -9.81" integrator="Euler" timestep="0.05"/>
```
What would be the equivalent of this line in `dm_control.composer` and where should this line go. I suspect it should be somewhere in the `FallOutOfView(composer.Task)` class definition but I don't know what the line is so I can't try it. 

2. As for velocities, I am not even sure how velocities of objects that are not `joints` (but `geom` like my ball) work in MuJoCo, since I have only ever seen velocity been observed directly from MuJoCo in the form of joint velocities indexed with the key/attribute `qvel`. I suspect to get my ball velocity as part of my state I would need to create a `sensor` which observes change in position and makes the velocity for the most recent transition observable but to try and test this I need to get my ball moving first which I don't know how to do without it having any joints.

ishank-juneja, 2022-01-25:
@vaxenburg Thank you for your help with my earlier question. Please take a look at this related question if you get a chance.

yuvaltassa, 2022-01-25:
1. The `mjcf_model` has `model.option` which contains `gravity` and all the other [options](https://mujoco.readthedocs.io/en/latest/XMLreference.html#option).
2. A geom cannot move. The things that move are bodies, and bodies can have child geoms which are fixed in the frame of the body. The way that bodies are allowed to move is defined by their joints, which specify how the body is allowed to move w.r.t its parent. Your geom is a child of the world and the world body which is immobile by definition. One way to add joints is to use the `attach` method. I recommend a quick read of the MuJoCo [overview chapter](https://mujoco.readthedocs.io/en/latest/overview.html).

closed 2022-01-25 by yuvaltassa

Coder-RG, 2022-02-25:
@yuvaltassa What is the correct way to create a `composer.Entity` from XML path, since the documentation only shows this using PyMJCF? Also, how do I access the actuator control value, while creating an observable? mjData has `ctrl` and `actuator_force` containing the same values. Here is some sample code.

```python
from dm_control import mujoco, composer, mjcf
from dm_control.composer.observation import observable

class QuadEnv(composer.Entity):
    def _build(self):
        self._model = mjcf.from_path('quad.xml')
    
    def _build_observables(self):
        return QuadObservables(self)
    @property
    def mjcf_model(self):
        return self._model
    
    @property
    def actuators(self):
        return tuple(self._model.find_all('actuator'))

class QuadObservables(composer.Observables):
    """Adds Sensor data and current actuator values as observables"""

    @composer.observable
    def rotor_values(self):
        rotors = self._entity.mjcf_model.find_all('actuator')
        return observable.MJCFFeature()

```