google-deepmind/dm_control #244: `group` named parameter in `worldbody.add` method

ishank-juneja, 2022-01-26:
From the PyMJCF tutorial:
```
for i, model in enumerate(creatures):
  # Place spawn sites on a grid.
  spawn_pos = (xpos.flat[i], ypos.flat[i], zpos.flat[i])
  spawn_site = arena.worldbody.add('site', pos=spawn_pos, group=3)
  # Attach to the arena at the spawn sites, with a free joint.
  spawn_site.attach(model).add('freejoint')
```
I couldn't find documentation for the `group` param of the `worldbody.add` method and my IDE was having trouble pointing me since `add` is a common method name. Would appreciate it if someone could explain the group parameter to me.

vaxenburg, 2022-01-26:
In this case, `group` is an attribute of `site`. See the [docs](https://mujoco.readthedocs.io/en/latest/XMLreference.html#body-site)

alimuldal, 2022-01-26:
Yep, the `mjcf` module thinly wraps MuJoCo's XML-based model description language. We do also have some documentation on `mjcf` itself [here](https://github.com/deepmind/dm_control/blob/master/dm_control/mjcf/README.md), but we don't attempt to cover the whole MJCF format there - for further info on that you should check out the MuJoCo docs @vaxenburg linked to above.

closed 2022-01-26 by alimuldal