google-deepmind/dm_control #245: How to install on google colab after forking the dm_control in your github account?

navidyou, 2022-01-28:
#This was the error I received after trying to install the fork or clone of dm_control stored in my github account 

ERROR: Command errored out with exit status 1: /usr/bin/python3 -c 'import io, os, sys, setuptools, tokenize; sys.argv[0] = '"'"'/content/src/dm-control/setup.py'"'"'; __file__='"'"'/content/src/dm-control/setup.py'"'"';f = getattr(tokenize, '"'"'open'"'"', open)(__file__) if os.path.exists(__file__) else io.StringIO('"'"'from setuptools import setup; setup()'"'"');code = f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();exec(compile(code, __file__, '"'"'exec'"'"'))' develop --no-deps Check the logs for full command output.

#this is the code I executed on google colab for installing the fork of dm_control stored in my github account 

`
##@title Run to install MuJoCo and `dm_control`
import distutils.util
import subprocess
if subprocess.run('nvidia-smi').returncode:
  raise RuntimeError(
      'Cannot communicate with GPU. '
      'Make sure you are using a GPU Colab runtime. '
      'Go to the Runtime menu and select Choose runtime type.')

mujoco_dir = "$HOME/.mujoco"

print('Installing OpenGL dependencies...')
!apt-get update -qq
!apt-get install -qq -y --no-install-recommends libglew2.0 > /dev/null

print('Downloading MuJoCo...')
MUJOCO_VERSION = 210
MUJOCO_ARCHIVE = (
    f'mujoco{MUJOCO_VERSION}-{distutils.util.get_platform()}.tar.gz')
!wget -q "https://mujoco.org/download/{MUJOCO_ARCHIVE}"
!wget -q "https://mujoco.org/download/{MUJOCO_ARCHIVE}.sha256"
check_result = !shasum -c "{MUJOCO_ARCHIVE}.sha256"
if _exit_code:
  raise RuntimeError(
      'Downloaded MuJoCo archive is corrupted (checksum mismatch)')

print('Unpacking MuJoCo...')
MUJOCO_DIR = '$HOME/.mujoco'
!mkdir -p "{MUJOCO_DIR}"
!tar -zxf {MUJOCO_ARCHIVE} -C "{MUJOCO_DIR}"

# Configure dm_control to use the EGL rendering backend (requires GPU)
%env MUJOCO_GL=egl

print('Installing dm_control...')
# TODO(b/211110746): Upgrade to the latest version once PyPI package is out.
#################!pip install -q dm_control==0.0.403778684

#instead of using deepmind account use the fork of dm_control in your account
!pip install -e git+https://github.com/navidyou/dm_control#egg=dm_control==0.0.403778684

print('Checking that the dm_control installation succeeded...')
try:
  from dm_control import suite
  env = suite.load('cartpole', 'swingup')
  pixels = env.physics.render()
except Exception as e:
  raise e from RuntimeError(
      'Something went wrong during installation. Check the shell output above '
      'for more information.\n'
      'If using a hosted Colab runtime, make sure you enable GPU acceleration '
      'by going to the Runtime menu and selecting "Choose runtime type".')
else:
  del suite, env, pixels
`

saran-t, 2022-01-31:
Could you please try again with the updated Colab notebook that was uploaded today?

saran-t, 2022-01-31:
Closing this issue. If the problem persists in the updated colab notebook please reopen.

closed 2022-01-31 by saran-t

navidyou, 2022-01-31:
Hello Saran, Thank you so much for your update. The package installs like before. However, I still cannot install a fork or clone of dm_control. Have you tried to fork or clone dm_control to a new github account and install it from there?

alimuldal, 2022-01-31:
Try removing the `-e` flag from your `pip install` invocation ("edit-mode" installations don't work because of the way we automatically generate the ctypes bindings for MuJoCo during the build process)

navidyou, 2022-01-31:
@alimuldal  Removing -e flag from pip install solved my problem. You saved me from a lot of headache. Thank you so so much 

I could install the clone of dm_control from my local github account. 

@saran-t I did have to use the new colab that was updated today where Mujoco was downloaded from deepmind rather than mujoco.org. Thank you so so much 

Thank you both for your help. I really appreciate it.