google-deepmind/dm_control #247: Cannot set physics timestep

markub3327, 2022-01-31:
Hello,

   i tried to set `<option timestep=".01"/>` in humanoid.xml, but I get an error about: 
> ValueError: Control timestep (0.025) must be an integer multiple of physics timestep (0.01)

I need to set 100Hz sampling rate on accelerometer and gyroscope placed on Humanoid's torso.

Thanks a lot.
Have a nice day.

yuvaltassa, 2022-01-31:
Have you tried changing the [control timestep](https://github.com/deepmind/dm_control/blob/master/dm_control/suite/humanoid.py#L30) so it's an integer multiple of the physics timestep? They can be equal of course.

closed 2022-01-31 by yuvaltassa

markub3327, 2022-01-31:
Sorry in documentation for MuJoCo I cannot see `control timestep` parameter. It's not so clear. 

I change it in code.

Thanks a lot.