google-deepmind/dm_control #248: What are units of accelerometer and gyroscope?

markub3327, 2022-01-31:
Hello,

   in documentation is written about accelerometer and gyroscope, but nothing about their units.

Is gyroscope in deg/s or rad/s?
Is accelerometer in m/s2?

What is the X, Y and Z axises orientation in `humanoid.xml` when is mounted on the torso? 
I need to compare it with the orientation of IMU in the smartphone.

Thanks a lot.
Have a nice day.

markub3327, 2022-01-31:
@yuvaltassa

yuvaltassa, 2022-02-02:
Re units: answered [here](https://github.com/deepmind/mujoco/issues/119).

The orientation is the same as the associated site. You can load the model in `simulate` and turn on site frame visualisation if you're not sure what it is. Specifically for the humanoid, neither the [parent body](https://github.com/deepmind/dm_control/blob/master/dm_control/suite/humanoid.xml#L33) nor [the site](https://github.com/deepmind/dm_control/blob/master/dm_control/suite/humanoid.xml#L38) have an orientation specification, the orientation will be aligned with the world frame.

closed 2022-02-02 by yuvaltassa