google-deepmind/dm_control #249: Cannot use global variable in viewer's policy function

markub3327, 2022-01-31:
Hello,

  I want to get an actual step of the run. I create a global variable called `pointer` under the policy function. The viewer gets me an error about unassigned variable:
> ERROR:absl:dm_control viewer intercepted an environment error.
> Original message: local variable 'pointer' referenced before assignment
> dm_control viewer intercepted an environment error.
> Original message: local variable 'pointer' referenced before assignment

Here is my code:
```
# state buffer
state_buffer = np.zeros((1, 300, 6), dtype=np.float32)
pointer = 0

# Define a uniform random policy.
def random_policy(time_step):
  action = np.zeros(action_spec.shape)

  acc = time_step.observation['imu'][0]
  gyro = time_step.observation['imu'][1]

  # FIFO
  if pointer == 300:
    plt.figure(figsize=(30, 10), dpi=80)
    for ax in range(3):
      yf = fft(state_buffer[0, :, ax])
      xf = fftfreq(300, 1.0 / SAMPLE_RATE)
      plt.subplot(4, 3, ax+1)
      plt.plot(xf, np.abs(yf))
    
      plt.subplot(4, 3, ax+4)
      plt.plot(state_buffer[0, :, ax])

      sos = butter(6, 6, 'lowpass', fs=SAMPLE_RATE, output='sos')
      gravity = sosfilt(sos, state_buffer[0, :, ax])

      plt.subplot(4, 3, ax+7)
      plt.plot(gravity)

      # remove gravity
      state_buffer[0, :, ax] -= gravity

      plt.subplot(4, 3, ax+10)
      plt.plot(state_buffer[0, :, ax])

    plt.show()

    plt.figure(figsize=(30, 10), dpi=80)
    for ax in range(3):
      yf = fft(state_buffer[0, :, ax+3])
      xf = fftfreq(300, 1.0 / SAMPLE_RATE)
      plt.subplot(2, 3, ax+1)
      plt.plot(xf, np.abs(yf))
    
      plt.subplot(2, 3, ax+4)
      plt.plot(state_buffer[0, :, ax+3])

    plt.show()
    
    pointer = 0

  input_data = np.concatenate([acc, gyro], axis=0)
  print(input_data)
  state_buffer[0, pointer] = input_data
  #print(state_buffer)

  # increment timestep
  pointer += 1

  return action

# Launch the viewer application.
viewer.launch(env, policy=random_policy)
```

kevinzakka, 2022-03-09:
You need to [explicitly](https://stackoverflow.com/a/423596) set the pointer variable to be global, i.e.,

```python
def random_policy(time_step):
    global pointer
```

markub3327, 2022-03-09:
Yes, working well. Thanks.

closed 2022-03-09 by markub3327