google-deepmind/dm_control #251: How to step composer.Environment in viewer?

chisarie, 2022-02-02:
Hi, 
I followed the example from chapter 5 of your paper: https://arxiv.org/pdf/2006.12983.pdf and everything seems to work fine. I now would like to open the environment in the viewer. What happens is that the viewer opens, but no steps are taken (I can see in the viewer that Time remains 0.0). Am I doing something wrong? I am calling the viewer like this:
```python
def random_policy(time_step):
    del time_step  # Unused.
    return np.random.uniform(low=-0.5, high=0.5, size=8)

creature = Creature(num_legs=4)
task = PressWithSpecificForce(creature)
env = composer.Environment(task)
viewer.launch(env, policy=random_policy)
```
Thanks!!

Just for reference, my full script is the following (copied from your paper):

```python
# The PyMJCF library provides a Python object model for MuJoCo's MJCF modelling
# language, which can describe complex scenes with articulated bodies. The goal of the
# library is to allow users to interact with and modify MJCF models programmatically
# using Python, similarly to what the JavaScript DOM does for HTML.

from dm_control import mjcf
from dm_control import composer
from dm_control import viewer
from dm_control.composer.observation import observable
from dm_control.composer import variation
from dm_control.composer.variation import distributions
from dm_control.composer.variation import noises
from dm_control.locomotion.arenas import floors
import numpy as np
import matplotlib.pyplot as plt
import cv2

BODY_RADIUS = 0.1
BODY_SIZE = (BODY_RADIUS, BODY_RADIUS, BODY_RADIUS / 2)

class Leg(object):
    """A 2-DoF leg with position actuators."""

    def __init__(self, length, rgba):
        self.model = mjcf.RootElement()
        # Defaults:
        self.model.default.joint.damping = 2
        self.model.default.joint.type = "hinge"
        self.model.default.geom.type = "capsule"
        self.model.default.geom.rgba = rgba
        # Thigh:
        self.thigh = self.model.worldbody.add("body")
        self.hip = self.thigh.add("joint", axis=[0, 0, 1])
        self.thigh.add("geom", fromto=[0, 0, 0, length, 0, 0], size=[length / 4])
        # Shin:
        self.shin = self.thigh.add("body", pos=[length, 0, 0])
        self.knee = self.shin.add("joint", axis=[0, 1, 0])
        self.shin.add("geom", fromto=[0, 0, 0, 0, 0, -length], size=[length / 5])
        # Position actuators:
        self.model.actuator.add("position", joint=self.hip, kp=10)
        self.model.actuator.add("position", joint=self.knee, kp=10)

def make_creature(num_legs):
    """Constructs a creature with 'num_legs' legs."""
    rgba = np.random.uniform([0, 0, 0, 1], [1, 1, 1, 1])
    model = mjcf.RootElement()
    model.compiler.angle = "radian"  # Use radians.
    # Make the torso geom.
    torso = model.worldbody.add(
        "geom", name="torso", type="ellipsoid", size=BODY_SIZE, rgba=rgba
    )
    # Attach legs to equidistant sites on the circumference.
    for i in range(num_legs):
[... 188 lines cut ...]
            random_state=random_state,
        )
        self._creature.set_pose(physics, position=creature_pose)
        self._button.set_pose(physics, position=button_pose)

    def get_reward(self, physics):
        return self._button.num_activated_steps / NUM_SUBSTEPS

def main():
    creature = Creature(num_legs=4)
    task = PressWithSpecificForce(creature)
    env = composer.Environment(task)

    # Define a uniform random policy.
    def random_policy(time_step):
        del time_step  # Unused.
        return np.random.uniform(low=-0.5, high=0.5, size=8)

    viewer.launch(env, policy=random_policy)
    return

if __name__ == "__main__":
    main()

```

kevinzakka, 2022-02-03:
Press the `space` bar to start/stop the physics. If you want to view the command palette, `F1` is your friend.

chisarie, 2022-02-04:
Ah, it was so simple! Thank you a lot Kevin :)

closed 2022-02-04 by chisarie