google-deepmind/dm_control #254: `logging` object is None?

zplizzi, 2022-02-16:
I get the following error after shutting down a training script using a `dm_control` environment with ctrl-c:

```
Exception ignored in: <function _create_finalizer.<locals>.callback at 0x7f2dc7839700>
Traceback (most recent call last):
  File "/opt/venv/lib/python3.9/site-packages/dm_control/mujoco/wrapper/core.py", line 262, in callback
TypeError: 'NoneType' object is not callable
Exception ignored in: <function _create_finalizer.<locals>.callback at 0x7f2dba8f3af0>
Traceback (most recent call last):
  File "/opt/venv/lib/python3.9/site-packages/dm_control/mujoco/wrapper/core.py", line 262, in callback
TypeError: 'NoneType' object is not callable
Exception ignored in: <function _create_finalizer.<locals>.callback at 0x7f2dba824c10>
Traceback (most recent call last):
  File "/opt/venv/lib/python3.9/site-packages/dm_control/mujoco/wrapper/core.py", line 262, in callback
TypeError: 'NoneType' object is not callable
```

When I look at that [line](https://github.com/deepmind/dm_control/blob/master/dm_control/mujoco/wrapper/core.py#L262), it seems like it must be the `logging` object that is None? Not sure how that could have happened though, I'm not using `absl` or any logger in any of my own code.

My versions are:
- dm_control: 0.0.425341097
- absl-py: 1.0.0

zplizzi, 2022-02-16:
Here's a longer section of logs, after adding `print(f"logger is: {logging}")` immediately preceding line 262 in `dm_control/mujoco/wrapper/core.py`.

```
(...many thousands of this line)
logger is: <module 'absl.logging' from '/opt/venv/lib/python3.9/site-packages/absl/logging/__init__.py'>
logger is: <module 'absl.logging' from '/opt/venv/lib/python3.9/site-packages/absl/logging/__init__.py'>
logger is: <module 'absl.logging' from '/opt/venv/lib/python3.9/site-packages/absl/logging/__init__.py'>
logger is: <module 'absl.logging' from '/opt/venv/lib/python3.9/site-packages/absl/logging/__init__.py'>
logger is: <module 'absl.logging' from '/opt/venv/lib/python3.9/site-packages/absl/logging/__init__.py'>
logger is: <module 'absl.logging' from '/opt/venv/lib/python3.9/site-packages/absl/logging/__init__.py'>
^CTraceback (most recent call last):
  File "/opt/projects/generally_intelligent/standalone/ballworld/quarantine/zack/dreamer/main.py", line 369, in <module>
    test_reward = test(args, models)
  File "/opt/projects/generally_intelligent/standalone/ballworld/quarantine/zack/dreamer/test.py", line 32, in test
    belief, posterior_state, action, next_observation, reward, done = update_belief_and_act(
  File "/opt/projects/generally_intelligent/standalone/ballworld/quarantine/zack/dreamer/stuff.py", line 27, in update_belief_and_act
    next_observation, reward, done = env.step(action.cpu() if isinstance(env, EnvBatcher) else action[0].cpu())
  File "/opt/projects/generally_intelligent/standalone/ballworld/quarantine/zack/dreamer/env.py", line 244, in step
    observations, rewards, dones = zip(*[env.step(action) for env, action in zip(self.envs, actions)])
  File "/opt/projects/generally_intelligent/standalone/ballworld/quarantine/zack/dreamer/env.py", line 244, in <listcomp>
    observations, rewards, dones = zip(*[env.step(action) for env, action in zip(self.envs, actions)])
  File "/opt/projects/generally_intelligent/standalone/ballworld/quarantine/zack/dreamer/env.py", line 122, in step
    state = self._env.step(action)
  File "/opt/venv/lib/python3.9/site-packages/dm_control/suite/wrappers/pixels.py", line 93, in step
    return self._add_pixel_observation(time_step)
  File "/opt/venv/lib/python3.9/site-packages/dm_control/suite/wrappers/pixels.py", line 110, in _add_pixel_observation
    pixels = self._env.physics.render(**self._render_kwargs)
  File "/opt/venv/lib/python3.9/site-packages/dm_control/mujoco/engine.py", line 211, in render
    image = camera.render(
  File "/opt/venv/lib/python3.9/site-packages/dm_control/mujoco/engine.py", line 860, in render
    ctx.call(self._render_on_gl_thread, depth=depth, overlays=overlays)
  File "/opt/venv/lib/python3.9/site-packages/dm_control/_render/executor/render_executor.py", line 136, in call
    return func(*args, **kwargs)
  File "/opt/venv/lib/python3.9/site-packages/dm_control/mujoco/engine.py", line 774, in _render_on_gl_thread
    mjlib.mjr_render(self._rect, self._scene.ptr,
KeyboardInterrupt

wandb: Waiting for W&B process to finish, PID 174502
wandb: Program failed with code 255.  Press ctrl-c to abort syncing.
wandb:
wandb: Find user logs for this run at: /opt/projects/generally_intelligent/standalone/ballworld/quarantine/zack/dreamer/wandb/run-20220216_141331-2tjm8mj3/logs/debug.log
wandb: Find internal logs for this run at: /opt/projects/generally_intelligent/standalone/ballworld/quarantine/zack/dreamer/wandb/run-20220216_141331-2tjm8mj3/logs/debug-internal.log
wandb: Run summary:
wandb:                episode 0
wandb:      environment_steps 2500
wandb:         training_steps 100
wandb:       observation_loss 90.39632
wandb:            reward_loss 0.23399
wandb:                kl_loss 3.01396
wandb:             actor_loss -3.26384
wandb:             value_loss 3.61483
wandb:          train_rewards 28.42377
wandb:   train_episode_length 499
wandb:               _runtime 48
wandb:             _timestamp 1645049659
wandb:                  _step 0
wandb: Run history:
wandb:                episode ▁
wandb:      environment_steps ▁
wandb:         training_steps ▁
wandb:       observation_loss ▁
wandb:            reward_loss ▁
wandb:                kl_loss ▁
wandb:             actor_loss ▁
wandb:             value_loss ▁
wandb:          train_rewards ▁
wandb:   train_episode_length ▁
wandb:               _runtime ▁
wandb:             _timestamp ▁
wandb:                  _step ▁
wandb:
wandb: Synced 5 W&B file(s), 0 media file(s), 0 artifact file(s) and 0 other file(s)
wandb:
wandb: Synced quadruped-walk: XXX
logger is: <module None from None>
Exception ignored in: <function _create_finalizer.<locals>.callback at 0x7fab5824b790>
Traceback (most recent call last):
  File "/opt/venv/lib/python3.9/site-packages/dm_control/mujoco/wrapper/core.py", line 263, in callback
TypeError: 'NoneType' object is not callable
logger is: <module None from None>
Exception ignored in: <function _create_finalizer.<locals>.callback at 0x7fab4b303b80>
Traceback (most recent call last):
  File "/opt/venv/lib/python3.9/site-packages/dm_control/mujoco/wrapper/core.py", line 263, in callback
TypeError: 'NoneType' object is not callable
logger is: <module None from None>
Exception ignored in: <function _create_finalizer.<locals>.callback at 0x7fab4b235ca0>
Traceback (most recent call last):
  File "/opt/venv/lib/python3.9/site-packages/dm_control/mujoco/wrapper/core.py", line 263, in callback
TypeError: 'NoneType' object is not callable
```

Somehow that logging object does become `module None from None`, idk what that means.

saran-t, 2022-03-15:
Sorry for the late response. There's a known issue in the deletion of the mjrContext object, which unfortunately still isn't fully fixed in our latest 1.0.0 release.

The error that you see should not prevent you from using `dm_control` normally, although I understand that it's a bit of a nuisance to see these messages popping up in your log.

closed 2022-03-15 by saran-t